stack-of-tasks/pinocchio #299: Stable API at the current status of devel branch

jmirabel, 2016-08-09:
Hi Pinocchio developpers,

The API of the devel branch is not stable enough for downstream libraries. Every week, something is changing in the API and I see a lot of issues which involves other API changes.

Would it be possible to have a branch which would be updated with the bug fixes but not all the API changes ? It could be the master branch (updated to the current devel branch) or another branch.

Thanks

jcarpent, 2016-08-09:
Currently, some methods or functions are set to deprecated, that means that they will be suppress in the near future. So, you have time to adapt your code.
During the meantime, the API is increased with new features. Then, they won't affect your code.

Concerning the Collision part, the API must be adapted to fulfil your requirements. Then it is hard to stabilise everything together.

jcarpent, 2016-08-09:
Also, if you want to directly discuss with Pinocchio developers, there is the email address pinocchio@laas.fr

jcarpent, 2016-08-09:
What I propose is to add a devel-stable or hop-stable branch inside the repo, which will be only filled with bug fixes. What do you think?

jmirabel, 2016-08-09:
Yep, a hpp-stable or devel-stable branch would be nice.

jcarpent, 2016-08-09:
Ok, I just pushed a new branch dedicated to the merge of Pinocchio inside HPP (https://github.com/stack-of-tasks/pinocchio/tree/hpp-devel). We'll try to maintain it up to date for bug fixes.

jcarpent, 2016-08-09:
Can we close this Issue?

closed 2016-08-09 by jmirabel