stack-of-tasks/pinocchio #301: Bugs in URDF parser

jmirabel, 2016-08-17:
I am currently fixing the URDF parser and there are a few points on which I'm not completely sure:

- [ ] https://github.com/stack-of-tasks/pinocchio/blob/devel/src/parsers/urdf/model.cpp#L392 the frame placement is different with and without inertia. Is this correct ?
- [ ] Zero inertias are not an issues for kinematic only algorithm, right ?
- [ ] Have the positions of geometries been check for geom in fixed joints, geom with offsets ?

jcarpent, 2016-08-17:
1. This is bug. It must be jointPlacement instead of nextPlacementOffset. Concerning this issue, we must think on: what is a frame of a body. Is it the frame centered around the centre of mass or the frame from which the inertia is defined?
What is the origin frame of the meshes?

jcarpent, 2016-08-17:
2. It is right. It is an issue only for ABA and the LLT decomposition.

jcarpent, 2016-08-17:
3.I do not get the question. Please rephrase.

jcarpent, 2016-08-17:
4.There is no package, only a urdf model in models directory with meshes.

jmirabel, 2016-08-17:
> This is bug. It must be jointPlacement instead of nextPlacementOffset. Concerning this issue, we must think on: what is a frame of a body. Is it the frame centered around the centre of mass or the frame from which the inertia is defined? What is the origin frame of the meshes?

The modifications I have added to the parser puts a body for each link of the URDF. As pinocchio do not define (yet?) any convention, I think taking what the file defines is the right behaviour.

> 3.I do not get the question. Please rephrase.

Sorry, I have updated the question.

> 4.There is no package, only a urdf model in models directory with meshes.

https://github.com/stack-of-tasks/pinocchio/blob/devel/unittest/geom.cpp#L256

jcarpent, 2016-08-17:
For the two last points, it is not my work. It was done by @fvalenza. Hope it can answer you rapidly.

jcarpent, 2016-08-17:
1.Becareful, when you call `appendBodyToJoint` it also adds a frame for the body automatically. 
In addition, I think this method must return the id of the created frame (because, in case of empty name, we cannot retrieve the body frame info).

jmirabel, 2016-08-17:
> In addition, I think this method must return the id of the created frame (because, in case of empty name, we cannot retrieve the body frame info).

:+1: it makes sense to me.

jcarpent, 2016-08-26:
This Issue seems to be solved. I'll close it now.

closed 2016-08-26 by jcarpent

jmirabel, 2016-08-26:
They have been corrected in the not yet merged pull-request #302