stack-of-tasks/pinocchio #305: Bug when python wrapper built

jcarpent, 2016-08-30:
There is currently one bug in the packaging of Pinocchio.
When one compiles with Python bindings, eigenpy is added as a Pinocchio dependency in the .pc file.
This addition of such a decency leads to crashes when calling Python Interpreter inside an application depending on Pinocchio. This bug is recently appeared in the SoT application.

To solve this bug, I suggest to split the Python bindings from the pinocchio package and create an independent python_pinocchio package (depending on pinocchio, boost python, Python and eigenpy only).

nim65s, 2016-08-30:
You'll need something like
`git filter-branch --prune-empty --subdirectory-filter src/python devel`

nmansard, 2016-08-30:
Yes! And it would make exactly 100 packages maintained by the team (we currently have 99 packages and desperately looking for the 100th). 
Could we possibly  think of an alternative?

jcarpent, 2016-08-30:
I know and I understand your pessimism. But, to say the truth, we'll never reach HPP ;)

Another solution is to remove eigenpy inside the packaging procedure. So, it may be necessary to add a cmake macro as `ADD_REQUIRED_BUILD_DEPENDENCY`which will not add the package to the .pc file.

jcarpent, 2016-09-02:
Any feedback?

olivier-stasse, 2016-09-21:
Another point why we *SHOULD* fix this: Working on our pattern generator I found out that with the current status of pc file we have to bind python to be able to use pinocchio. This does not make sense.

olivier-stasse, 2016-09-21:
A possible solution would be to remove from pinocchio.pc the boost python component, the required dependency to eigenpy and generate by other means than the cmake modules an other .pc file.

olivier-stasse, 2016-09-21:
I have a possible solution implementing the above solution here:
https://github.com/olivier-stasse/pinocchio/tree/devel
One problem was coming the parser-python which IMHO should be move to libpinocchio_pywrap.
I still have a problem when doing make tests (parser-python is failing)

fvalenza, 2016-09-21:
@olivier-stasse
For the make test fails did you install the python bindings before (ie make install then make test)?

jcarpent, 2016-09-21:
I already implemented a fix for that issue. I will sync the devel branch accordingly ASAP.

jcarpent, 2016-09-21:
I reopened an old issue concerning the division of the pinocchio module into several packages. As the project becomes bigger, it seems necessary to divide pinocchio from its bindings. I would suggest to create a new repo python_pinocchio or pinocchiopy (name suggested by @olivier-stasse) containing the python bindings and having its own cmake project.

aelkhour, 2016-09-21:
I've been loosely following this thread.

Regarding @jcarpent's proposal, I fear it could make life harder for both developers and users, as they will always have to make sure they have first built and installed the correct commit/version of pinocchio before attempting to build pinocchio bindings.

It seems as if the only problem here comes from the pkg-config file, and @olivier-stasse's proposal of having two .pc files (same as for the libs) seems to address this issue.

@jcarpent Do you foresee any other issues that would justify separating pinocchio into multiple repos?

jcarpent, 2016-09-21:
I see your points.
But it can really help the packaging into debian package for example.

jcarpent, 2016-09-27:
Solved by #333 and #337.

closed 2016-09-27 by jcarpent