stack-of-tasks/pinocchio #308: Bug in JointModelPlanar integrate and/or difference

jmirabel, 2016-09-02:
The following code gives an example where `q0 + (q1 - q0)` is different from `q1`.

The following code:

```cpp
#include "pinocchio/spatial/explog.hpp"
#include "pinocchio/multibody/joint/joint-planar.hpp"
#include "pinocchio/math/quaternion.hpp"

int main(int, char**)
{
  using namespace se3;

  typedef JointModelPlanar::ConfigVector_t CV;
  typedef JointModelPlanar::TangentVector_t TV;
  typedef JointModelPlanar::Transformation_t SE3;

  JointModelPlanar jmodel;
  jmodel.setIndexes(0,0,0);

  CV q0, q1, q;
  double u = 0;

  q0 << -3.1267286, -3.337924, 3.01237;
  q1 << -3.2      , -4       , 0      ;
  u = 0.9983735;
  q = jmodel.interpolate(q0, q1, u);

  std::cout << "q0: "<< q0.transpose() << std::endl;
  std::cout << "q1: "<< q1.transpose() << std::endl;
  std::cout << "q1-q0: "<< jmodel.difference(q0,q1).transpose() << std::endl;
  std::cout << "q0 + (q1-q0): "<< jmodel.integrate (q0,jmodel.difference(q0,q1)).transpose() << std::endl;
  std::cout << "u : "<< u << std::endl;
  std::cout << "q : "<< q.transpose() << std::endl;
}
```

gives the following output:
```
q0: -3.12673 -3.33792  3.01237
q1: -3.2   -4    0
q1-q0:    1.09052 -0.0497723   -3.01237
q0 + (q1-q0): -3.04717 -2.61905        0
u : 0.998374
q :   -3.04894   -2.61897 0.00489962
```

I couldn't find the reason myself... Any idea of what is wrong ?

nmansard, 2016-09-02:
jmodel.integrate  (q0,jmodel.difference(q1,q0)) ?

jmirabel, 2016-09-02:
I understood the issue, which was appearing when the angular velocity was negative. I will submit a fix.

jcarpent, 2016-09-02:
Nice !

jcarpent, 2016-09-05:
Solved by #309.

closed 2016-09-05 by jcarpent