stack-of-tasks/pinocchio #312: API of joint configuration methods in JointModelBase derived class

jmirabel, 2016-09-06:
The current API of `JointModelBase` for `integrate`, `difference`, `interpolate` and `distance` takes as input configurations and velocities having the size of the robot model, instead of the size of the joint. The second option is better because:
- design problem (questionable): it would be both more elegant and less code. The joints would not need to know their position and there would be not need for long lines like
```cpp
AnUnreadableTypeCorrespondingToASegment q_0 = q.segment<NQ>(idx_q());
```
The unreadable type would be guessed by the compiler.
- currently, `interpolate` cannot call `integrate` and `distance` cannot call `difference` the formers do not know the size of the robot velocity.

jcarpent, 2016-09-06:
:+1:
I'm in favour of the second option too.

nmansard, 2016-09-09:
BTW (talking about API), these newly-added functions are adding some more lines in classes that already were quite heavy (both in the API and number of implementation lines). I would be in favor of bringing them out of the class.

1. API with functions
call se3::joint::normalize(jointModel,q) rather than jointModel.normalize(q)

2. By moving the code
we may want to create new files in src/multibody/joint/manifold/joint-free-flyer.hxx where the implementation of the methods are done. It could also be added in src/algorithm/joint/free-flyer.hxx, with a src/algorithm/joint.hpp where the definition of the template functions is done and the joint/*.hxx files are included.

Both 1. and 2. are independent. I am not sure of any of them.

3. Should we add a mandatory forwardKinematics method in all joints (ie is API JointModelBase), that calc would call?

jcarpent, 2016-09-09:
This is my opinion concerning the update of the API.

1. I would say, we need these methods for each joint model which are independent to the position of the joint in the kinematic tree. Such methods must appear in the Joint class model and will only take argument of dimension Joint::ConfigVector and TangentVector. This is minimalist and mandatory if we want to work at the joint level only.
Then, we need to add algo which unroll for kinematic model and call the corresponding methods. 
This will help the adding of new joints. It won't be necessary to update algo.

2. I would go for a details directory like that  src/multibody/joint/{details,impl} (manifold seems not appropriate as naming).

3. I think FK is mandatory as the code is duplicate in calc(q) and calc(q,v) and can be used for other purposes, like integrate methods for example.

nmansard, 2016-09-12:
Just to be sure to properly understand.
1. You think that we must keep class methods (i.e. jmodel.normalize(q_i)) rather than friend functions (i.e. normalize(jmodel,q_i)). Correct?
My point is that these methods are marginally important compared to e.g. forwardKinematics. We could tolerate a joint without normalize for example, as long as the user is not going to call normalize with this joint. Everything should be able to compile like that. This fact would be clearer if the functions is not part of the standard joint API.

2. {details,impl} are too generic. For the same reason exposed upper, I would prefer to make clear that the corresponding code is marginal wrt. e.g. calc and FK. Not really important.

3. Lets do it.

jcarpent, 2016-09-30:
This issue will be solved along with #344.

jmirabel, 2018-10-03:
I think all issues have been addressed. Please, re-open is not.

closed 2018-10-03 by jmirabel