stack-of-tasks/pinocchio #313: Conflicting frame names

jmirabel, 2016-09-07:
I have a robot with link and joint having the same name. In Pinocchio, it causes a conflict in the Frame vector because the frames are identified by their names.

I see 2 possible solutions:
- the very ugly one, which I do not like at all: prepending "joint/" before joint names and / or "body/" before body names.
- the more elegant one: identifying frames by their name and their type. This means the API must be updated: replace all `existBodyName`, `getBodyId`, `getFrameId`... by 2 methods only `existFrame(name, type)` and `getFrameId(name, type)` with possibly a default value for type (which I do not like.

The second option simplyfies the API and is a lot more elegant so I'm strongly favoring it.

jmirabel, 2016-09-07:
The argument `type` of function `existFrame` and `getFrameId` could even be a mask. For instance, to check if a frame of type `JOINT` or `FIXED_JOINT` exists, one could do:
```cpp
model.existFrame (frameName, JOINT | FIXED_JOINT)
```
This means the enum `FrameType` values must be set manually.

nmansard, 2016-09-09:
Solved by #302

closed 2016-09-09 by nmansard