stack-of-tasks/pinocchio #317: SRDF parser not tested.

jmirabel, 2016-09-08:
There is not unittest for the SRDF parser. Maybe it could be added to `unittest/urdf.cpp`.

jcarpent, 2016-09-08:
Split if you can, it will be easier to maintain. I prefer several unit test files than a big one.

jmirabel, 2016-09-08:
The suggested file is merely loading a URDF, and is less than 10 lines of actual code. To test the SRDF, the easiest is to first load a URDF. So you will anyway copy the same 10 lines of code.

jcarpent, 2016-09-08:
Yes, and we will add new parsing of infos, like end effectors, etc ... and then the test number will increase with time. So, to avoid future refactoring, I propose to directly add such a dedicated file.

jcarpent, 2016-09-09:
As this is not mandatory, I remove the label.

jcarpent, 2016-09-27:
Solved by #342.

closed 2016-09-27 by jcarpent