stack-of-tasks/pinocchio #321: jacobian versus getJacobian

nmansard, 2016-09-13:
We have two methods doing similar computation:
1. computeJacobians then getJacobian pre-compute all possible joint jacobians (using a forward strategy) then return a specific jacobian. 
2. jacobian compute a single jacobian (using a backward strategy.

Issues:
A. The documentation is not sufficient. If my understanding is correct, it must be clear from the doc that the result of jacobian is the same as getJacobian (possibly specifying the arguments to get the same results).
B. Do we have any benchmarks about how faster jacobian is? It would be good to mention the ration in the doc.
C. Data::iMf must be better documented. What is the letter "f" standing for? Is it plan to use iMf otherwhere? The fact that iMf is updated (and not oMi as classically) in jacobian() should be mentionned in the doc.

jcarpent, 2016-09-20:
This is a partial answer.

B. In the unit test jacobian.cpp, you already implemented a simple benchmark to evaluate the cost of computeJacobians + getJacobian vs jacobian.

C. f stands for the frame index and i for the index of the joint.

jcarpent, 2018-10-03:
This issue has been solved by new release 1.3.0

closed 2018-10-03 by jcarpent