stack-of-tasks/pinocchio #323: Conversion from signed to unsigned types

fvalenza, 2016-09-15:
With the change of return type of methods such as addFrame, we often have such lines:
`FrameIndex jointFrameId = model.addJointFrame(idx, parentFrameId);`

with addJointFrame that returns int and FrameIndex of type std::size_t. As these methods now can return `-1`,  problems can rise (affecting -1 to unsigned long).

Shall we rethink the use of std::size_t as indexes ?

nmansard, 2016-09-15:
Yes we should revise the types of indexes in Pinocchio. I pled that 
unsigned int should never be used. BTW, some namings of indexes should 
also be changed (njoints instead of njoint, nframes instead of nFrames, 
or anything else to be consistent). We also need to discuss whether 
these variables are needed.

However, the reason you raised is not correct. It is a bad procedure to 
return -1 in case of failure. Either throw or assert. In that case, I 
would say assert, as the design pattern of Pinocchio is to assert in 
kernel functions, and throw in user functions. Here we should assert in 
model, and possibly do the check in the URDF parser, and there throw 
before calling the asserted function.

We also discussed that -1 can be used for "no information", for 
non-mandatory attributes (like parentFrame).

jmirabel, 2016-09-15:
The old prototype returned a `bool` saying whether there was an error. This wasn't the correct behaviour either.

jmirabel, 2016-09-15:
Sorry, my fingers got lost on the keyboard.

The current prototype originates for the old one, which was returning a `bool` telling whether is succeeded. I wanted to keep that behaviour.

I agree with forcing the use of `existFrame` before `addFrame`. I don't have a very strong opinion on whether to assert or throw. Isn't `addFrame` more a user function than a kernel function ? The user may want, a posteriori, to add some frames to the model, no?

closed 2018-07-02 by jcarpent