stack-of-tasks/pinocchio #327: hpp-devel branch No more compatible with eigen 3.0.5

fvalenza, 2016-09-16:
eigen3.0.5 is the default version (apt..) on Ubuntu 12.04.
Some changes in hpp-devel branch have calls to functions `hasNaN()` that are not in eigen 3.0.5 ( it is in 3.2.1, i don't know about versions between these two).

The solutions i see:
- Are these calls mandatory ? May be we can use something else ?
- Change RequiredDependency from eigen3 >= 3.0.5 to eigen3 >= 3.2.1 ( or may be less )

jmirabel, 2016-09-17:
`M.hasNaN()` can be replaced by `!((M.array()==M.array()).all())`.

This is from `Eigen3/src/Core/BooleanRedux.h`:
```cpp
template<typename Derived>
inline bool DenseBase<Derived>::hasNaN() const
{
  return !((derived().array()==derived().array()).all());
}
```

fvalenza, 2016-09-17:
Thanks for this answer.

https://github.com/fvalenza/pinocchio/commit/4169891e3950d0bceaf5623215e8441b8cf65275 should be part of my next PR. Is it ok ?

I have put it in details namespace so that the line `xxx.hasNaN() && yy.hasNaN()` is not too long

nmansard, 2016-09-20:
Yes it is a good idea to add your commit to the PR. Please also accout for @jcarpent remarks on your commit.

jcarpent, 2016-09-20:
Seems to be fixed.

closed 2016-09-20 by jcarpent