stack-of-tasks/pinocchio #328: in devel and hpp-devel branch assert when norm of quaternion not close to 1

florent-lamiraux, 2016-09-18:
In src/multibody/joint/joint-free-flyer.hpp: 216, the norm of the quaternion is checked against 1, but the tolerance is very small 1.5e-8 on my machine.

This makes my application crash in debug mode upon call to various methods like integrate or diff.
Did someone experience the same issue ?

jmirabel, 2016-09-18:
I encountered it due to repeated integration in the SE3 joint. I had to add
function "firstOrderNormalize" and call it when integrating quaternions of
SE3.
I didn't pay attention to the threshold but I think only the squared norm
can safely be checked with this precision. The first order normalization is
made on the squared norm and we cannot hope to have a 1e-16 precision with
a first order normalization...

Could you give more details about where is the assert in Pinocchio ?

jcarpent, 2016-09-19:
This should be removed. Or for numerical issues like the integration, we may need to add a compilation flag for that purpose, not an assert.

jcarpent, 2016-09-27:
Ping. @jmirabel modified the content of the assert replacing `quat.coeffs().norm()`with `quat.coeffs().squaredNorm()`. @florent-lamiraux Can you try again on your computer with the modifications?

florent-lamiraux, 2016-09-29:
It still fails when integrating several times with the GUI, although the angular velocity is equal to 1.
Maybe setting the threshold to 1e-4 is enough. The goal of this assert is not to check the accuracy of the quaternion norm, but to detect quaternions that are not normalized at all by mistake.

jcarpent, 2016-09-30:
Solved by #347.

closed 2016-09-30 by jcarpent