stack-of-tasks/pinocchio #336: Header dependency

jmirabel, 2016-09-22:
In third party library, when I include `pinocchio/multibody/joint/joint.hpp` before `pinocchio/multibody/joint/joint-composite.hpp`, I get a long unreadable error the first message of which is posted below.
I have no error if I include `pinocchio/multibody/joint/joint-composite.hpp` first.

```
include/pinocchio/multibody/joint/joint-composite.hpp: In constructor ‘std::vector<T, Eigen::aligned_allocator<T> >::vector(std::vector<T, Eigen::aligned_allocator<T> >::size_type, const value_type&) [with T = se3::JointModel; std::vector<T, Eigen::aligned_allocator<T> >::size_type = long unsigned int; std::vector<T, Eigen::aligned_allocator<T> >::value_type = se3::JointModel]’:
include/pinocchio/multibody/joint/joint-composite.hpp:126:43: error: invalid use of incomplete type ‘std::vector<se3::JointModel, Eigen::aligned_allocator<se3::JointModel> >::value_type {aka struct se3::JointModel}’
```

jmirabel, 2016-09-22:
The issue can be reproduced by including `joint.hpp` before `joint-composite.hpp` in `unittests/joint-composite.cpp`.

fvalenza, 2016-09-22:
You must include joint-composite.hpp before joint.hpp

The reason is currently jointcomposite has a member that is std::vector<Joint> so that he must know completely Joint, that itself must know completely jointocmposite due to jointocmposite being inside the variant.
This can, in the future be solved by changing in joint-ocmposite the std::vector<Joint> by std::vector<jointvariant>

jcarpent, 2016-09-22:
One solution consists in changing the std::vector of Joint{Model,Data} with Joint{Model,Data}Variant, to avoid circular inclusion. 
Also, we need to delete the global EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION for all the vector types in the se3 namespace. We need to properly specify them in each classes needing std::vector of  classes containing Eigen objects.

jmirabel, 2016-09-22:
I like the idea of deleting the global `EIGEN_DEFINE_STL_VECTOR_SPECIALIZATION`.

jcarpent, 2016-09-27:
Solved by #340.

closed 2016-09-27 by jcarpent