stack-of-tasks/pinocchio #338: JointIndex values

olivier-stasse, 2016-09-22:
Hi,
Sorry for this stupid question but I tried to check all the JointIndex values from 
models/simple_humanoid.urdf
For this I modified a test in value.hpp, the code is here:
https://github.com/olivier-stasse/pinocchio/blob/topic/JointIndexPb/unittest/value.cpp#L193

The result is the following:
model.parents.size():32
model.njoints:32
l_wrist:24
r_wrist:38
model.parents.size():31
model.njoints:31
CHEST idTest:31
WAIST_P idTest:27
WAIST_R idTest:29
LARM_WRIST_R idTest:45
LARM_WRIST_P idTest:43
LARM_WRIST_Y idTest:41
LARM_ELBOW idTest:39
LARM_SHOULDER_Y idTest:37
LARM_SHOULDER_R idTest:35
LARM_SHOULDER_P idTest:33
LLEG_ANKLE_R idTest:13
LLEG_ANKLE_P idTest:11
LLEG_KNEE idTest:9
LLEG_HIP_Y idTest:7
LLEG_HIP_P idTest:5
LLEG_HIP_R idTest:3
RARM_WRIST_R idTest:59
RARM_WRIST_P idTest:57
RARM_WRIST_Y idTest:55
RARM_ELBOW idTest:53
RARM_SHOULDER_Y idTest:51
RARM_SHOULDER_R idTest:49
RARM_SHOULDER_P idTest:47
RLEG_ANKLE_R idTest:25
RLEG_ANKLE_P idTest:23
RLEG_KNEE idTest:21
RLEG_HIP_Y idTest:19
RLEG_HIP_P idTest:17
RLEG_HIP_R idTest:15

You can see that r_wrist and several other joints are above the number of joints.
Is it normal ?
If yes, then how can we use the field model.parent ?

Thanks for your time and your help.
Olivier.

jcarpent, 2016-09-23:
In fact, you asked the id of the frame attached to the joint and having the same name. You should retrieve the Frame from model.frames[id] and get the parent with model.frames[i].parent (which corresponds to the parent joint).

jcarpent, 2016-09-23:
@olivier-stasse told me that the problem has been solved.

closed 2016-09-23 by jcarpent