stack-of-tasks/pinocchio #344: Remove integrate, distance, differentiate from JointModel classes and add TopologyBase class

jcarpent, 2016-09-29:
To be flexible and extensible, those methods need to be modifiable. The idea is to move them to a dedicated classes templated by the JointType having those methods.

jmirabel, 2016-12-20:
I propose to decide the name of the base class and of the file where it should go. I propose `LieGroupOperationBase` with a one file `src/multibody/liegroup/name` per file in file `src/multibody/joint/name`.

nmansard, 2016-12-20:
See also the same discussion in #312 

I think "manifold" would fit better than "Lie group"  (less idiomatic and at least as accurate, possibly more). I think Justin mentioned some name that seemed more suitable (ie easier to understand) like differential-topology, or something similar. 
I would prefer to have it as a subdirectory of joint, ie src/multibody/joint/manifold/manifold-operation-base.hpp. 

In any case, namings is easy to change. If you are in a good mood, go ahead with writing the code with the name you like. We can refine it afterwards if we later manage to all agree on a better name.

jcarpent, 2016-12-20:
LieGroup seems a complex term, certainly without real meaning for most users. Few weeks ago, I proposed something like Diff{efferential}Calculus but others propositions look nice too.

florent-lamiraux, 2016-12-20:
LieGroup is more restrictive than manifold, but as pinocchio inheritence is static, it is unlikely that we try to represent more general manifolds. I think the most general structure that we can represent with the current architecture is Riemanian Manifold, but it is a bit long.

jcarpent, 2016-12-20:
@florent-lamiraux. Use classname identifiable for any user -> we must avoid complex names as `Riemanian Manifold`even they have the right meaning.

jcarpent, 2017-01-06:
Solved by #365. Nevertheless, a refactoring of the code of #365 is needed to avoid duplicate codes between Lie Group operations and Spatial classes.

closed 2017-01-06 by jcarpent

jmirabel, 2017-01-06:
The code could be improved with the missing `inline` keyword.