stack-of-tasks/pinocchio #362: viewing mesh with scale parameter in gepetto-viewer

proyan, 2016-12-12:
This might be a bug. If a scale is specified in the urdf file for the meshes, this scale is not transmitted to gepetto-viewer for display.

Meshes are added to the viewer by `self.viewer.gui.addMesh(meshName, meshPath)` in robot_wrapper.py. This does not take into account any additional information available in the urdf. There should ideally be a way to transmit the full information of the urdf to the viewer.

jmirabel, 2016-12-12:
From the viewer side, there is a function setScale. So if Pinocchio keeps
scale information, the fix should be easy.

jcarpent, 2016-12-13:
@proyan You should update the class `GeometryObject` with a scale parameter and keeps this information when parsing the URDF file. That's pretty easy and straight forward.

jmirabel, 2016-12-13:
If you do so, please make it explicit that the scale has already been taken into account. I mean:
`GeometryObject.meshPath` + `GeometryObject.scale` -> `GeometryObject.fcl`

proyan, 2016-12-13:
In case of other objects being loaded, (e.g. Cylinder/Box/sphere), should the default scale be set to (1,1,1)?

jcarpent, 2016-12-13:
What should be the other solution you have in mind?

proyan, 2016-12-13:
#363 Pull request created

jcarpent, 2016-12-16:
Solved by #363.

closed 2016-12-16 by jcarpent