stack-of-tasks/pinocchio #373: Runtime error from boost::bad_get

Pouya-moh, 2017-02-14:
Hello,

I have to apologize in advance since this will probably turns out to be an error of mine and not an issue of Pinocchio. I am looking around for an alternative to our current humanoid computation library (translation: I am very new to Pinocchio!). I tried to do a very simple test to get familiar with the way of pinocchio which is like this:

```
se3::Model model;
se3::Data  data(model);
se3::urdf::buildModel("path/to/model.urdf",model);
Eigen::VectorXd q = Eigen::VectorXd::Random(model.nq);
Eigen::Vector3d com = se3::centerOfMass(model,data,q);

```

This compiles, however, there is a runtime error:
```
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::bad_get> >'
  what():  boost::bad_get: failed value get using boost::get
```
Debugger traces it back to [these calls](https://gist.github.com/Pouya-moh/acc3a571d7bc54c9fe51d3db81a0c98c#file-gistfile1-txt). In another attempt I tried another functionality (`se3::crba(model,data,q);` to do a `se3::getComFromCrba(model,data);`) but same error with [this gist](https://gist.github.com/Pouya-moh/5a50457c7ba45431c44b1cf12901d474#file-gistfile1-txt).

Any idea about what am I doing wrong here? Thanks!
<hr>
<sub>Built from source.<br>Built with -DBUILD_PYTHON_INTERFACE=OFF<br>Tested with both master and devel branch</sub>

jcarpent, 2017-02-14:
Hi Pouya,

Before investigating your issue, can you tell me which version of Boost, Eigen and urdfdom are you using?

Justin

jcarpent, 2017-02-14:
Ok, your first need to create your model and then to build your data structure.
```
se3::Model model;
se3::urdf::buildModel("path/to/model.urdf",model); // first load your model ...
se3::Data data(model); // ... and then build data from model
Eigen::VectorXd q = Eigen::VectorXd::Random(model.nq);
Eigen::Vector3d com = se3::centerOfMass(model,data,q);
```

Best,

Pouya-moh, 2017-02-14:
Hi jcarpent,
You are indeed right. Thank you very much.

Best.

closed 2017-02-14 by Pouya-moh