stack-of-tasks/pinocchio #375: spurious output

florent-lamiraux, 2017-02-15:
Would it be possible not to pollute the standard output when using pinocchio ?
If necessary https://github.com/humanoid-path-planner/hpp-util is a light weight package that provides logging into files.

jcarpent, 2017-02-15:
Discussing with @jmirabel, I got your issue. We can indeed use LOG system. I would go for boost-log to avoid any further dependency.

But it seems that you obtain some outputs from verbosity mode when reading URDF model. This behavior is expected, isn't it?

So, we can also stream to LOG variables.

jmirabel, 2017-02-23:
IMHO, `verbose` should not be an arguments of functions, like in prototype of `buildModel`. Verbosity should be controlled project-wise (through macros or free-functions) and not function-wise.

jcarpent, 2017-02-23:
I think it really depends on your application and what you want to see. Sometines, one just needs local printed info for a specific function, like when one loads an URDF file.

jmirabel, 2017-02-23:
The functionality you talk about can be achieved in many other ways.

- Function-wise control pollutes function prototype and do not add extra functionality.
- The output of `buildModel` is very long and has to be written to a file for inspection. It wouldn't be less readable with other stuff in the file.
- Project-wise control can be made very easy to parse. This is easier than modifying some c++ code to enable verbosity.

jcarpent, 2017-02-23:
Ok, I'm open to propostions or pointees: what do you suggest to do?
The main idea is to avoid big files with thousands of lines.

jcarpent, 2017-02-23:
In addition, we want the log data be available from Python.

jmirabel, 2017-02-23:
> In addition, we want the log data to be available from Python.

Boost Python makes `std::cout` available in Python ? If you work with stream, you can redirect the stream wherever you want.

> The main idea is to avoid big files with thousands of lines.

In the first tutorial of HPP, Pinocchio prints 3308 lines to the standard output... I don't think you can avoid files with thousands of lines...

jmirabel, 2017-02-23:
My suggestion is either internal macros (you can probably get some inspiration from hpp-util) or external library like boost-log or glog.

jcarpent, 2019-02-13:
@florent-lamiraux Would you have time to solve this issue?

jcarpent, 2020-01-13:
With no activity for a while, I will close this issue.

closed 2020-01-13 by jcarpent