stack-of-tasks/pinocchio #376: Compiling from source

sheim, 2017-02-22:
Hi,  
I managed to install pinocchio a couple weeks ago from source on macOS 10.12, but today I've been trying to reinstall from source to keep all SoT repos in the same place, without success. I used brew to install `boost-python` (1.63.0) and `eigen` (3.3.2), and installed `eigenpy`, `urdfdom_headers` from SoT directly, and `urdfdom` from ros, but I get errors in the `pinocchio/src/parsers/urdf/` folder (see error log).

Any ideas what the issue is?

[errorlog.txt](https://github.com/stack-of-tasks/pinocchio/files/793336/errorlog.txt)

jcarpent, 2017-02-22:
Hi,
The issues comes from the fact your version of urdfdom uses std::shared_ptr instead older boost::shared_ptr. std::shared_ptr does not provide a comparision against classic NULL pointer but rather std::nullptr which is a C++11 feature. 
Normally, #377 should solve your issue.

Concerning the Cmake Warning, it concerns only developers. It is just an info to mention that developers have to take care of iinkage against boost_python, nothing more. You can suppress this info by using the Cmake option `-Wno-dev`.

Best

sheim, 2017-02-23:
Thanks, but no dice. And I tried comparing the stderr from master branch and devel branch, but the actual errors all seem to persist in the devel branch. Seems to be a problem casting types from {boost, std} to urdf types.  
As a temporary fix, could you also point me to the version of urdf that SoT is working with?
[errLog_devel.txt](https://github.com/stack-of-tasks/pinocchio/files/796175/errLog.txt)

jcarpent, 2017-02-23:
Ok, I'm fixing the complete bug comming from URDFDOM >= 0.4.2.
If you need a stable version, please use URDFDOM 0.4.1. 

Anyway, do you want to use the SoT on OS X?

sheim, 2017-02-23:
Yeah, that would be great : )

jcarpent, 2017-02-23:
For sure, if you wanna work with Ubuntu, the installation on 14.04 is straightforward using binaries. So, my recommendation is to use a virtual machine with ubuntu 14.04.
The instructions are here: http://stack-of-tasks.github.io/download.html

Otherwise, if you absolutely want to work under OS X, there is a partial brew repo available at https://github.com/jcarpent/homebrew-gepetto

sheim, 2017-02-23:
That's super!  
brew install pinocchio failed on a broken (404) link: `https://git.openrobots.org/projects/robotpkg-wip/repository/revisions/master/raw/pinocchio/patches/patch-aa` which I'm guess just got moved somewhere else. (and possibly to patch-ab also)

If it's not something trivial though I'll move to ubuntu (it's more of a convenience matter).

jcarpent, 2017-02-24:
Normally, the compatibility with last version 1.0.0 of URDFDOM has been fixed through #378.
I also update https://github.com/jcarpent/homebrew-gepetto repo with last release of Pinocchio 1.2.3.

Hopping everything goes well with your installation.

sheim, 2017-02-24:
Yes, installs well on both brew and from source now, thanks!

Incidentally, I also installed the gepetto-viewer (from source), but I can't load the gepetto.corbaserver in ipython as per the tutorial. I'm looking into this now, and will otherwise switch to ubuntu. Just to confirm, [this](https://github.com/humanoid-path-planner/gepetto-viewer) is still the correct gepetto-viewer to use (and not the [gepetto-corba-viewer](https://github.com/humanoid-path-planner/gepetto-viewer-corba) from hpp)?

jcarpent, 2017-02-24:
the viewer is composed of two distinct parts, the engine contained in gepetto-viewer (an overlayer on top of OSG) and gepetto-viewer-corba which is the server for gepetto-viewer. 
In other words, we launch a corba-server through gepetto-viewer-corba (hpp-gui on OS X) and we are able to use it either in Python or C++. Corba is used as a middleware to communicate between the applications (HPP or Pinocchio) and the viewer.

nmansard, 2017-02-27:
You likely want to use Gepetto viewer through its corba interface 
(that's how it is done in the tutorials using Python). Hence you need 
both packages.

sheim, 2017-03-01:
Okay. I had some trouble getting gepetto-viewer-corba to compile, namely the brew installation of `open-scene-graph` doesn't include osgQt. So I compiled this from source, from [here](https://github.com/openscenegraph/osgQt/commit/930c7ee20be6b5c1c344c3c53a75bd05032f1d71): had to roll back because the brew open-scene-graph version is still on 3.5.5 (and the HEAD of osgQt on that repo requires 3.5.6, but only since a few commits back).  

With this, I can now `import gepetto.corbaserver` in ipython. However, launching `gepetto-viewer-server` (or `gepetto-gui`) throws an exception with `CORBA::NO_RESOURCES`, as in [hpp/gepetto-viewer-corba#42](https://github.com/humanoid-path-planner/gepetto-viewer-corba/issues/42), however the solutions there didn't work for me (I suspect from inconsistencies between linux and macOS). Haven't gone into detail yet but I have to stop for today so I thought I'd post an quick update.

Also, I installed on ubuntu 14.04, and thought I'd note a couple things:
- in the [download page](http://stack-of-tasks.github.io/pinocchio/download.html), the instruction  
`sudo echo "deb [arch=amd64] http://robotpkg.openrobots.org/packages/debian/pub trusty robotpkg" >> /etc/apt/sources.list.d/robotpkg.list`  
doesn't work since sudo is only being applied to `echo` but not to `>>`, so it should be changed to  
`echo "deb [arch=amd64] http://robotpkg.openrobots.org/packages/debian/pub trusty robotpkg" | tee --append /etc/apt/sources.list.d/robotpkg.list`  
(see [here](http://stackoverflow.com/questions/84882/sudo-echo-something-etc-privilegedfile-doesnt-work-is-there-an-alterna))
- I could launch `gepetto-gui` (solved the same issue above following the instructions for omniNames), but had none of the examples were available to be loaded. Haven't checked out how come.
- I could not import the `gepetto` module in python. When compiling with CMake, cmake  
```
Could NOT find PythonQt (missing QtAll)  
skipping PythonQt settings
```  
but I haven't managed to solve this yet.

jmirabel, 2017-03-01:
Some help on the viewer side.

---
> With this, I can now import gepetto.corbaserver in ipython

seems to contradict with:
> I could not import the gepetto module in python.

You can safely ignore the `PythonQt` warning. Check your `PYTHONPATH`.

------
> throws an exception with CORBA::NO_RESOURCES, as in hpp/gepetto-viewer-corba#42, however the solutions there didn't work for me

You tried:
- `sudo apt-get install omniorb-nameserver`
- `sudo omniNames -start`

sheim, 2017-03-01:
Sorry for the confusion: so the first part is in **macOS** (can import, having issues with omniNames, will look into it later), the second part is in **ubuntu 14.04** (cannot import, `sudo omniNames -start` works).

jmirabel, 2017-03-01:
I understand. The most likely is an incomplete `PYTHONPATH`. Could you give the output of:
```bash
echo $PYTHONPATH
locate gepetto/corbaserver/__init__.py
```

sheim, 2017-03-02:
I get  
```
> echo $PYTHONPATH
/opt/openrobots/lib/python2.7/site-packages
> sudo updatedb
> locate gepetto/corbaserver/__init__.py
/home/sheim/Repositories/SoT/gepetto-viewer-corba/_build/src/gepetto/viewer/corba/gepetto/corbaserver/__init__.py
/home/sheim/Repositories/SoT/gepetto-viewer-corba/src/gepetto/corbaserver/__init__.py
/usr/local/lib/python2.7/site-packages/gepetto/corbaserver/__init__.py
```

jmirabel, 2017-03-02:
Does the following command fix the issue ?
```bash
export PYTHONPATH=${PYTHONPATH}:/usr/local/lib/python2.7/site-packages/
```

jcarpent, 2017-03-11:
@sheim Are you done with your issue?

sheim, 2017-03-11:
Sorry, was working on other stuff. @jmirabel yes that fixes things in ubuntu, thanks!

closed 2017-03-11 by sheim

tristan-ka, 2017-11-17:
Hi,

I am trying to install pinocchio on macOS. I am using the https://github.com/jcarpent/homebrew-gepetto repo and it seems that I cannot access ftp://trac.laas.fr/pub/openrobots/eigenpy/eigenpy-1.3.1.tar.gz to download the eigenpy module. Does it come from my connection or is it located elsewhere now ?

Regards

jcarpent, 2017-11-17:
Thanks for this issue. It appears that you may be locked by your institution's network.
I've just updated the remote addresses on https://github.com/jcarpent/homebrew-gepetto for LAAS's formulas.
You may try again.

tristan-ka, 2017-11-17:
Thank you very much for being so reactive