stack-of-tasks/pinocchio #379: Add option to reorder joint after parsing URDF

thomasfla, 2017-02-28:
To be compatible with other dynamic libraries or robot drivers, it would be good to be able to choose joint order after parsing URDF file, taking as input a vector of joint names.

Even if to be efficient Pinocchio needs to place parents before their child, there is still freedom in ordering the sub-chains of the tree. If the reordering asked by user doesn't respect this parent-child constraint, Pinocchio could report an error.

This need isn't a priority as memory map is possible, but so far it has been express by T. KO, G. BUONDONNO and myself.

nmansard, 2017-02-28:
I think it is a good idea. An algorithm could be implemented to re-order 
Model (no need to reorder Data, that would be re-created from the new 
model after reordering). It would also be necessary to re-order 
GeomModel, although it can be implemented in a second part of the work. 
I do not expect any blocking difficulty in implementing it, although it 
is a tedious work to have all the indexes correct afterward. I could 
supervise the implementation. Who could lead the task?

jmirabel, 2017-02-28:
For the information, the only reason why the joints are reordered is that `urdfdom` stores them in a map, which is an ordered set. They are sorted by joint name. IMHO, this is a mistake in `urdfdom`...

For the record, the problem is in `Model::initTree`, which should not loop on a map but on a vector. The fix is very easy here, but it will be a pain to get the change accepted by the package maintainers...

jcarpent, 2017-02-28:
@jmirabel The request of the feature is not only due to URDF choice of traversing tree, but depends on the target application. For instance, with the HRP-2, the joint order is not the same in the VRML file than in the URDF file. Then, you may want that in Model, the joint ordering be the same than in old application to avoid unnecessary mapping.

cmastalli, 2018-11-23:
Joint desired order could be described through a SDF.

tlemarec, 2021-02-26:
Hello, any news on this issue ?

jcarpent, 2021-02-26:
Sorry, but we are terribly lacking in manpower. Your contribution @tlemarec will be really appreciated for implementing this feature!

tlemarec, 2021-02-26:
I'll see what we can do (@davidfauthoux), our understanding of the inner workings of pinocchio is still limited

jcarpent, 2021-02-26:
Normally, this should not be so hard, knowing we already support the build of a reduced model from a given model.
For efficiency reasons, you just need to have the correct ordering, meaning that the index of a parent is always below that the ones of its children + for a given subtree, the indexes should be successive.

york-fu, 2025-09-06:
Pinocchio determines the joint order through depth-first traversal but arranges sibling chains alphabetically at branch points, which affects the order of q. When loading a URDF file, the URDF file is structured from top to bottom, naturally defining one chain first and then another, clearly specifying the order of sibling chains at branch points. Why not follow this order for arrangement?

york-fu, 2025-09-06:
Pinocchio determines the joint order through depth-first traversal. Due to the use of the map container, sibling chains are arranged alphabetically at branch points, which affects the order of q. When loading a URDF file, the file is structured from top to bottom. Naturally, one chain is defined first, followed by another, explicitly specifying the order of sibling chains at branch points. Why not follow this order for arrangement?