stack-of-tasks/pinocchio #380: cmake: missing header in v1.2.3

nim65s, 2017-03-27:
```
In file included from pinocchio-1.2.3/bindings/python/multibody/fcl/expose-fcl.cpp:19:0:
pinocchio-1.2.3/build/include/pinocchio/bindings/python/multibody/fcl/contact.hpp:21:59: erreur fatale : pinocchio/spatial/fcl-pinocchio-conversions.hpp : Aucun fichier ou dossier de ce type
 #include "pinocchio/spatial/fcl-pinocchio-conversions.hpp"
```
It should be added here: https://github.com/stack-of-tasks/pinocchio/blob/v1.2.3/CMakeLists.txt#L160

jcarpent, 2017-03-28:
I do not have any trouble with the release 1.2.3. Can you tell more about your issue?

nim65s, 2017-03-28:
I can't reproduce… Sorry about that.

proyan, 2017-03-29:
Same problem in Kevin's system as well. There is no installation of HPP-FCL, but it is apparently not removed inside the bindings. I think the header should be conditional on the availability of Hpp-fcl. Right?

Current version: 1.2.3
System: 14.04, 64-bit

`In file included from /usr/HRP2/pinocchio/bindings/python/multibody/fcl/expose-fcl.cpp:19:0:
/usr/HRP2/pinocchio/_build_release/include/pinocchio/bindings/python/multibody/fcl/contact.hpp:21:59: fatal error: pinocchio/spatial/fcl-pinocchio-conversions.hpp: No such file or directory
 #include "pinocchio/spatial/fcl-pinocchio-conversions.hpp"`

nim65s, 2017-03-29:
It's weird… But yes, the header is already conditionally available: 
https://github.com/stack-of-tasks/pinocchio/blob/v1.2.3/CMakeLists.txt#L272

I tried to recompile without HPP-FCL and it worked… I'll try in a clean container tonight.

proyan, 2017-03-29:
> It's weird… But yes, the header is already conditionally available:
https://github.com/stack-of-tasks/pinocchio/blob/v1.2.3/CMakeLists.txt#L272

The problem is in python bindings `pinocchio/bindings/python/multibody/fcl/contact.hpp`
If I understand correctly, the files in the bindings/python/multibody/fcl folder are for exposing fcl objects, so they should not be included at all

jcarpent, 2017-03-31:
Solved by #383.

PR #383 (2017-03-31): Handle FCL exposition if hpp-fcl is not found
Must fix #380.
closed 2017-03-31 by jcarpent