stack-of-tasks/pinocchio #381: urdf 1.0.0 not supported

proyan, 2017-03-29:
urdfdom 1.0.0 seems to give errors with declarations of the shared_pointers from urdf
Below is the example. This error is duplicated for all the urdf types (urdf::collision, urdf::material etc)
`In file included from /usr/HRP2/pinocchio/_build_release/include/pinocchio/parsers/urdf.hpp:25:0,
                 from /usr/HRP2/pinocchio/src/parsers/urdf/model.cpp:20:
/usr/HRP2/pinocchio/_build_release/include/pinocchio/parsers/urdf/types.hpp:44:30: error: conflicting declaration ‘typedef class boost::shared_ptr<urdf::Box> urdf::BoxSharedPtr’
   URDF_TYPEDEF_CLASS_POINTER(Box);
                              ^
/usr/HRP2/pinocchio/_build_release/include/pinocchio/parsers/urdf/types.hpp:38:32: note: in definition of macro ‘URDF_TYPEDEF_CLASS_POINTER’
 typedef URDF_SHARED_PTR(Class) Class##SharedPtr; \
                                ^
In file included from /usr/HRP2/install/include/urdf_model/joint.h:44:0,
                 from /usr/HRP2/install/include/urdf_model/link.h:44,
                 from /usr/HRP2/install/include/urdf_model/model.h:42,
                 from /usr/HRP2/pinocchio/_build_release/include/pinocchio/parsers/urdf/types.hpp:21,
                 from /usr/HRP2/pinocchio/_build_release/include/pinocchio/parsers/urdf.hpp:25,
                 from /usr/HRP2/pinocchio/src/parsers/urdf/model.cpp:20:
/usr/HRP2/install/include/urdf_model/types.h:53:1: error: ‘urdf::BoxSharedPtr’ has a previous declaration as ‘typedef class std::shared_ptr<urdf::Box> urdf::BoxSharedPtr’
 URDF_TYPEDEF_CLASS_POINTER(Box);
 ^
In file included from /usr/HRP2/pinocchio/_build_release/include/pinocchio/parsers/urdf.hpp:25:0,
                 from /usr/HRP2/pinocchio/src/parsers/urdf/model.cpp:20:
/usr/HRP2/pinocchio/_build_release/include/pinocchio/parsers/urdf/types.hpp:44:30: error: conflicting declaration ‘typedef class boost::shared_ptr<const urdf::Box> urdf::BoxConstSharedPtr’
   URDF_TYPEDEF_CLASS_POINTER(Box);
                              ^
/usr/HRP2/pinocchio/_build_release/include/pinocchio/parsers/urdf/types.hpp:39:38: note: in definition of macro ‘URDF_TYPEDEF_CLASS_POINTER’
 typedef URDF_SHARED_PTR(const Class) Class##ConstSharedPtr; \
                                      ^
In file included from /usr/HRP2/install/include/urdf_model/joint.h:44:0,
                 from /usr/HRP2/install/include/urdf_model/link.h:44,
                 from /usr/HRP2/install/include/urdf_model/model.h:42,
                 from /usr/HRP2/pinocchio/_build_release/include/pinocchio/parsers/urdf/types.hpp:21,
                 from /usr/HRP2/pinocchio/_build_release/include/pinocchio/parsers/urdf.hpp:25,
                 from /usr/HRP2/pinocchio/src/parsers/urdf/model.cpp:20:
/usr/HRP2/install/include/urdf_model/types.h:53:1: error: ‘urdf::BoxConstSharedPtr’ has a previous declaration as ‘typedef class std::shared_ptr<const urdf::Box> urdf::BoxConstSharedPtr’
 URDF_TYPEDEF_CLASS_POINTER(Box);
 ^
In file included from /usr/HRP2/pinocchio/_build_release/include/pinocchio/parsers/urdf.hpp:25:0,
                 from /usr/HRP2/pinocchio/src/parsers/urdf/model.cpp:20:
/usr/HRP2/pinocchio/_build_release/include/pinocchio/parsers/urdf/types.hpp:44:30: error: conflicting declaration ‘typedef class boost::weak_ptr<urdf::Box> urdf::BoxWeakPtr’
   URDF_TYPEDEF_CLASS_POINTER(Box);
                              ^
/usr/HRP2/pinocchio/_build_release/include/pinocchio/parsers/urdf/types.hpp:40:30: note: in definition of macro ‘URDF_TYPEDEF_CLASS_POINTER’
 typedef URDF_WEAK_PTR(Class) Class##WeakPtr
                              ^
In file included from /usr/HRP2/install/include/urdf_model/joint.h:44:0,
                 from /usr/HRP2/install/include/urdf_model/link.h:44,
                 from /usr/HRP2/install/include/urdf_model/model.h:42,
                 from /usr/HRP2/pinocchio/_build_release/include/pinocchio/parsers/urdf/types.hpp:21,
                 from /usr/HRP2/pinocchio/_build_release/include/pinocchio/parsers/urdf.hpp:25,
                 from /usr/HRP2/pinocchio/src/parsers/urdf/model.cpp:20:
/usr/HRP2/install/include/urdf_model/types.h:53:1: error: ‘urdf::BoxWeakPtr’ has a previous declaration as ‘typedef class std::weak_ptr<urdf::Box> urdf::BoxWeakPtr’
 URDF_TYPEDEF_CLASS_POINTER(Box);`

olivier-stasse, 2017-03-29:
Dear Rohan,
On which distribution are you trying to compile pinocchio ?

jcarpent, 2017-03-29:
which branch?

proyan, 2017-03-29:
We were on:
Ubuntu 14.04 64-bit
devel branch

Installation of urdf 0.4.2 fixed this error.

olivier-stasse, 2017-03-29:
Please stick to the official release of the package.
urdfdom is 0.2.10 on 14.04 LTS
If we start to mix software from packages distribution and pull from the repository we will run into troubles.

proyan, 2017-03-29:
Noted

closed 2017-03-29 by proyan