stack-of-tasks/pinocchio #384: Add options to separate algorithms

proyan, 2017-04-12:
Currently, some algorithms are duplicated in others. For example:
- computeJacobians duplicates forwardKinematics
- jacobianCenterOfMass duplicates computeJacobian (but explicitly asks for forwardKinematics)
- ccrba duplicates crba

Would it be possible to explicitly specify the duplicated algorithm in the argument list to avoid recomputations? For example, computeJacobians could also take a parameter asking for forwardKinematics, and so on

jcarpent, 2017-04-12:
computeAllTerms do all the computation once.
You should use this one if your need all the dynamics term.

If you really want to avoid internal recomputations, the idea is to write a personalize algo by yourself.

proyan, 2017-04-12:
> computeAllTerms do all the computation once.
You should use this one if your need all the dynamics term.

That would mean additional computations even if they are not needed. My basic argument is to avoid these.

> If you really want to avoid internal recomputations, the idea is to write a personalized algo by yourself.

In that case, I would be reimplementing the same algorithms already done in pinocchio, which beats the purpose.

I did the jacobian time test:
Over hundred iterations, current computeJacobians algorithm takes 2.9 us, and we add a flag for forward kinematics, the time is 3.1us for computeJacobians(forwardKinematics=true)

jcarpent, 2017-04-12:
The idea is to compute once all the terms and then to call for example getJacobian to recover the desired Jacobian from the current values in data.

proyan, 2017-04-12:
Currently the call to computeJacobians _implies_ a call to forwardKinematics. In my opinion, it would be better to make this call explicit.

closed 2018-07-03 by jcarpent