stack-of-tasks/pinocchio #385: Pinocchio does not compile with Eigen 3.3

andreadelprete, 2017-04-21:
Pinocchio doesn't seem to compile with Eigen 3.3, which is the last stable release of Eigen. This doesn't allow me to benefit from the last developments of Eigen, such as the implementation of the [Complete Orthogonal Decomposition](https://eigen.tuxfamily.org/dox/classEigen_1_1CompleteOrthogonalDecomposition.html).

jcarpent, 2017-04-22:
I checked a bit, but it seems that the problem comes from humanoid-path-planner/hpp-fcl#10 where @jmirabel added by default the use of Eigen vectors inside FCL instead of internal linear algebra library.

jcarpent, 2017-04-22:
To tackle this issue, you can set the flag FCL_EIGEN to OFF when installing hpp-fcl.

andreadelprete, 2017-04-22:
Actually after some tests I found out that the problem was that I had not recompiled pinocchio with Eigen 3.3.3, so I in my library I was mixing Eigen 3.2 (used to compile pinocchio) with Eigen 3.3.3. After recompiling pinocchio everything seems to work fine.

jcarpent, 2017-04-22:
Ok. Perfect. I close the issue then.

closed 2017-04-22 by jcarpent