stack-of-tasks/pinocchio #389: Porting from pinocchio 1.1.0 to 1.2.3

aelkhour, 2017-05-26:
I am attempting to port our codebase from pinocchio 1.1.0 to 1.2.3. The pinocchio documentation has been enhanced since, but I could not find porting notes, so I will list my questions here as I go through this process.

## Retrieving fix body transforms and using them.

I would like to port the following code:

```c++
// Get position with respect to last moving parent
se3::SE3 const transformInParentFrame = model_->fix_lmpMi[bodyIndexIn];
// Retrieve last moving parent index
int32_t const lastMovingParentIndex = model_->fix_lastMovingParent[bodyIndexIn];
// Get last moving parent transform
se3::SE3 const parentTransformInWorld = data_->oMi[lastMovingParentIndex];
// Compute selected body transform
globalTransform = (parentTransformInWorld * transformInParentFrame);
```

After digging a little bit, I found out that `Model::fix_lmpMi` and `Model::fix_lastMovingParent` have disappeared, and that the notion of frames has been introduced. I would like how to compute the above placements using the new API.

jcarpent, 2017-05-29:
Hello Antonio, 

I do not have any memory of `Model::fix_lmpMi` and `Model::fix_lastMovingParent`. But I can help you to move from the old API to the new one.

So, the idea of FRAMEs is to avoid to directly use the notion of JOINTs for the end user. A FRAME can be a BODY, a SENSOR, a JOINT itself (there is a redundancy) and a FIXED JOINT. 
A FRAME has a flag (enum) for the type (BODY,JOINT,etc).
It has a direct parent (another FRAME) and the former parent inside the JOINT tree. 
And of course a name.
There are dedicated algo that work on FRAMEs (for instance framesForwardKinematics that updates the placement of the frames according to the former joint parent placement - you access to this information through data.oMf).
So, for your purpose, you just need to call framesForwardKinematics and then get your globalTransform through data.oMf[myFrameId].

aelkhour, 2017-05-29:
Thanks @jcarpent.

The algorithm `se3::framesForwardKinematics` seems to only do a forward geometry. If the frames are supposed to be part of the user API, it would make sense to have access to the frame spatial velocities and accelerations. Is this already available?

nmansard, 2017-05-29:
Spatial velocity and acceleration are the same than the one of the 
attached body. For example, nu_j = nu_body (with j=joint and b=body the 
two quantities you want to measure, and nu_x the spatial velocity of 
frame x). Here, nu_x is a geometrical object (not a numerical one) and 
is therefore note expressed in a particular frame. Now, you may want to 
express nu_x in the frame x, which I would denote by $^x{nu}_x$ (not 
very convenient in plain text), the velocity of x expressed in frame x. 
Then you would simply change the frame where nu_joint (or equivalently 
nu_body) is expressed:

$$^b{nu}_b = ^bX_j ^j{nu}_b = ^bX_j ^j{nu}_j$$

Note also that the velocity of b and j expressed in world frame o are 
exactly the same $^o{nu}_j = ^o{nu}_b$. Is that clear?

aelkhour, 2017-05-29:
Understood @nmansard, thanks.

My concern is regarding the retrieval of placement, velocities, accelelerations, etc. of a specific frame using the frame API. As you pointed out, I could rely on the joint velocities and accelerations provided by `Data::v` and `Data::a`, but they are vectors which are indexed by a joint id. I did not know how to retrieve the joint id from the frame, but I just realized that @jcarpent already explained that in the previous comment.

Assuming that I called the computation algorithms for a given state (q, v, a), I would then write something like:

```c++
// Check that frame exists in model.
if (model.existFrame(frameName))
{
  // Get frame index.
  se3::Model::FrameIndex const frameId = model.getFrameId(frameName);
  // Get current frame placement in world frame.
  se3::SE3 const& framePlacement = data.oMf[frameId];
  // Get frame parent joint id.
  se3::Model::JointIndex const& parentJointId = model.frames[frameId].parent;
  // Get current frame velocity in world frame.
  se3::Motion const& frameVelocity = data.v[parentJointId];
  // Get current frame acceleration in world frame.
  se3::Motion const& frameAcceleration = data.a[parentJointId];
}
```

Is this correct?

nmansard, 2017-05-29:
Yes it is. Although you evidently have to call the forward kinematics 
methods beforehand. And remember that frameVelocity (resp acc) is the 
velocity of the body frame expressed in joint frame.

aelkhour, 2017-05-29:
> And remember that frameVelocity (resp acc) is the
velocity of the body frame expressed in joint frame.

Fixed, thanks!

aelkhour, 2017-05-31:
It seems that porting was sucessful, so I'm closing this for now. Thanks for you help @jcarpent and @nmansard !

closed 2017-05-31 by aelkhour