stack-of-tasks/pinocchio #396: installation problem on ubuntu16.04

longwoo, 2017-07-27:
```
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
Failed to find Doxygen, documentation will not be generated.
-- eigen3 >= 3.0.5 is required.
-- Checking for module 'eigen3 >= 3.0.5'
--   Found eigen3 , version 3.2.92
-- Pkg-config module eigen3 v3.2.92 has been detected with success.
-- metapod >= 1.0.7 is optional.
-- Checking for module 'metapod >= 1.0.7'
--   
-- urdfdom >= 0.2.0 is optional.
-- Checking for module 'urdfdom >= 0.2.0'
--   Found urdfdom , version 0.4.0
-- Pkg-config module urdfdom v0.4.0 has been detected with success.
-- hpp-fcl >= 0.4.2 is optional.
-- Checking for module 'hpp-fcl >= 0.4.2'
--   
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   python
--   unit_test_framework
CMake Warning (dev) at cmake/boost.cmake:90 (MESSAGE):
  The boost_python library has been removed from the Boost_LIBRARIES
  variable.  To link againt boost_python, please use the macro
  TARGET_LINK_BOOST_PYTHON.
Call Stack (most recent call first):
  CMakeLists.txt:146 (SEARCH_FOR_BOOST)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- eigenpy >= 1.3.1 is required.
-- Checking for module 'eigenpy >= 1.3.1'
--   
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
  cmake/pkg-config.cmake:282 (PKG_CHECK_MODULES)
  cmake/pkg-config.cmake:490 (ADD_DEPENDENCY)
  bindings/python/CMakeLists.txt:57 (ADD_COMPILE_DEPENDENCY)

-- Configuring incomplete, errors occurred!
```

I follow the install from source instruction, but gives me the error when using ```cmake ..```

olivier-stasse, 2017-07-27:
According to the output you did not install eigenpy.

longwoo, 2017-07-27:
Thank you. It works.

olivier-stasse, 2017-07-27:
Glad it worked. Out of curiosity why did you not try the binary installation ?

longwoo, 2017-07-27:
I'v tried the binary installation, but some depends problems like this
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 robotpkg-pinocchio : Depends: robotpkg-hpp-fcl (= 0.4.2) but it is not going to be installed or
                               robotpkg-hpp-fcl (= 0.4.2r2) but it is not going to be installed or
                               robotpkg-hpp-fcl (= 0.4.2r1) but it is not going to be installed
                      Depends: robotpkg-eigenpy (= 1.3.2) but it is not going to be installed or
                               robotpkg-eigenpy (= 1.3.3) but it is not going to be installed or
                               robotpkg-eigenpy (= 1.3.1r1) but it is not going to be installed or
                               robotpkg-eigenpy (= 1.3.1) but it is not going to be installed
                      Depends: libboost-filesystem1.54-dev (>= 1.34.1) but it is not installable
                      Depends: libboost-filesystem1.54.0 (>= 1.34.1) but it is not installable
                      Depends: libboost-python1.54-dev (>= 1.34.1) but it is not installable
                      Depends: libboost-system1.54-dev (>= 1.34.1) but it is not installable
                      Depends: libboost-system1.54.0 (>= 1.34.1) but it is not installable
                      Depends: libboost-test1.54-dev (>= 1.34.1) but it is not installable
                      Depends: libboost-test1.54.0 (>= 1.34.1) but it is not installable
E: Unable to correct problems, you have held broken packages.
```
So I turned to source installation.

jcarpent, 2017-07-27:
In fact, currently our binary support do not install system dependencies like libboost-dev-all and so on. You must install them by yourself according to the output you have shown. 
Some people here @ LAAS-CNRS are working on that issue.

If you want to learn more on Pinocchio, there is also some tutorials available at http://projects.laas.fr/gepetto/index.php/Teach/InsaRobotics2016

longwoo, 2017-07-27:
Thanks. It would be very useful.  
By the way, since it is said that Pinocchio support C++ and python, is there any C++  example ? I plan to write some plugins in Gazebo.

olivier-stasse, 2017-07-27:
@jcarpent : Actually the issue has been solved and this is why the system is asking to install libboost.
That is very strange because according to https://packages.ubuntu.com/trusty/libs/libboost-test1.54.0
the packages should have installed.

@longwoo : Could you try
`
apt-get install libboost-filesystem1.54.0
`
and give us the output ?

olivier-stasse, 2017-07-27:
@longwoo:
Regarding the C++ example, you can find simple samples in the unit test directory:
https://github.com/stack-of-tasks/pinocchio/tree/master/unittest

longwoo, 2017-07-27:
@olivier-stasse here is the result:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package libboost-filesystem1.54.0 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libboost-filesystem1.54.0' has no installation candidate
```

jcarpent, 2017-07-27:
@longwoo Good initiative to support Pinocchio in Gazebo. As mentioned by @olivier-stasse, most of the examples are already included inside the unit tests, for instance the loading of a URDF inside unittest/urdf.cpp

olivier-stasse, 2017-07-27:
@longwoo : Thanks, then could you try
`apt-cache search libboost-filesystem`

jcarpent, 2017-07-27:
This is because the official version of boost is 1.58 on 16.04

longwoo, 2017-07-27:
@olivier-stasse The result:
```
libboost-filesystem1.58.0 - filesystem operations (portable paths, iteration over directories, etc) in C++
libboost-filesystem-dev - filesystem operations (portable paths, iteration over directories, etc) in C++ (default version)
libboost-filesystem1.58-dev - filesystem operations (portable paths, iteration over directories, etc) in C++

```

olivier-stasse, 2017-07-27:
@jcarpent: Good point
@longwoo : Thanks, Could you check that 
`/etc/apt/sources.list.d/robotpkg-sot.list`
is pointing to xenial (16.04) and not to trusty (14.04)

longwoo, 2017-07-31:
@olivier-stasse 
I did the process like this:
1.```lsb_release -c``` , it gives me ```Codename:	xenial```
2.```sudo sh -c "echo 'deb [arch=amd64] http://robotpkg.openrobots.org/packages/debian/pub  xenial robotpkg' >> /etc/apt/sources.list.d/robotpkg.list"```
3.```curl http://robotpkg.openrobots.org/packages/debian/robotpkg.key |
sudo apt-key add -```, 
it gives me ``` % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current  Dload  Upload   Total   Spent    Left  Speed 100   862  100   862    0     0    421      0  0:00:02  0:00:02 --:--:--   421 OK  ```
4. The point is here , when ```sudo apt-get update``` , it gives me the error 
```
Get:1 http://security.ubuntu.com/ubuntu xenial-security InRelease [102 kB]     
Hit:2 http://packages.osrfoundation.org/gazebo/ubuntu-stable xenial InRelease  
Hit:3 http://cn.archive.ubuntu.com/ubuntu xenial InRelease                     
Hit:4 http://ppa.launchpad.net/dartsim/ppa/ubuntu xenial InRelease             
Ign:5 http://robotpkg.openrobots.org/packages/debian/pub trusty InRelease      
Get:6 http://cn.archive.ubuntu.com/ubuntu xenial-updates InRelease [102 kB]    
Ign:7 http://ppa.launchpad.net/fcl-debs/ppa/ubuntu xenial InRelease            
Ign:8 http://robotpkg.openrobots.org/packages/debian/pub xenial InRelease      
Ign:9 http://mirrors.ustc.edu.cn/ros/ubuntu xenial InRelease                   
Get:10 http://security.ubuntu.com/ubuntu xenial-security/main amd64 Packages [316 kB]
Hit:11 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu xenial InRelease   
Get:12 http://robotpkg.openrobots.org/packages/debian/pub trusty Release [2,259 B]
Get:13 http://mirrors.ustc.edu.cn/ros/ubuntu xenial Release [3,807 B]          
Get:14 http://robotpkg.openrobots.org/packages/debian/pub xenial Release [2,259 B]
Get:15 http://mirrors.ustc.edu.cn/ros/ubuntu xenial Release.gpg [181 B]        
Get:16 http://security.ubuntu.com/ubuntu xenial-security/main i386 Packages [298 kB]
Ign:17 http://ppa.launchpad.net/libccd-debs/ppa/ubuntu xenial InRelease        
Get:18 http://robotpkg.openrobots.org/packages/debian/pub trusty Release.gpg [181 B]
Get:19 http://security.ubuntu.com/ubuntu xenial-security/main amd64 DEP-11 Metadata [59.2 kB]
Get:20 http://mirrors.ustc.edu.cn/ros/ubuntu xenial/main amd64 Packages [438 kB]
Get:21 http://security.ubuntu.com/ubuntu xenial-security/main DEP-11 64x64 Icons [54.2 kB]
Hit:22 http://ppa.launchpad.net/webupd8team/sublime-text-3/ubuntu xenial InRelease
Get:23 http://robotpkg.openrobots.org/packages/debian/pub xenial Release.gpg [181 B]
Get:24 http://security.ubuntu.com/ubuntu xenial-security/universe amd64 DEP-11 Metadata [40.6 kB]
Get:25 http://security.ubuntu.com/ubuntu xenial-security/universe DEP-11 64x64 Icons [56.1 kB]
Get:26 http://cn.archive.ubuntu.com/ubuntu xenial-backports InRelease [102 kB] 
Ign:27 http://ppa.launchpad.net/fcl-debs/ppa/ubuntu xenial Release             
Get:28 http://robotpkg.openrobots.org/packages/debian/pub trusty/robotpkg amd64 Packages [289 kB]
Ign:29 http://ppa.launchpad.net/libccd-debs/ppa/ubuntu xenial Release          
Ign:30 http://ppa.launchpad.net/fcl-debs/ppa/ubuntu xenial/main amd64 Packages 
Ign:31 http://ppa.launchpad.net/fcl-debs/ppa/ubuntu xenial/main i386 Packages  
Ign:32 http://ppa.launchpad.net/fcl-debs/ppa/ubuntu xenial/main all Packages   
Ign:33 http://ppa.launchpad.net/fcl-debs/ppa/ubuntu xenial/main Translation-en_US
Get:34 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 Packages [593 kB]
Get:35 http://robotpkg.openrobots.org/packages/debian/pub xenial/robotpkg amd64 Packages [272 kB]
Ign:36 http://ppa.launchpad.net/fcl-debs/ppa/ubuntu xenial/main Translation-en 
Get:37 http://mirrors.ustc.edu.cn/ros/ubuntu xenial/main i386 Packages [339 kB]
Ign:38 http://ppa.launchpad.net/fcl-debs/ppa/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:39 http://ppa.launchpad.net/fcl-debs/ppa/ubuntu xenial/main DEP-11 64x64 Icons
Ign:40 http://ppa.launchpad.net/libccd-debs/ppa/ubuntu xenial/main amd64 Packages
Ign:41 http://ppa.launchpad.net/libccd-debs/ppa/ubuntu xenial/main i386 Packages
Ign:42 http://ppa.launchpad.net/libccd-debs/ppa/ubuntu xenial/main all Packages
Ign:43 http://ppa.launchpad.net/libccd-debs/ppa/ubuntu xenial/main Translation-en_US
Get:44 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main i386 Packages [572 kB]
Ign:45 http://ppa.launchpad.net/libccd-debs/ppa/ubuntu xenial/main Translation-en
Ign:46 http://ppa.launchpad.net/libccd-debs/ppa/ubuntu xenial/main amd64 DEP-11 Metadata
Get:47 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main Translation-en [239 kB]
Ign:48 http://ppa.launchpad.net/libccd-debs/ppa/ubuntu xenial/main DEP-11 64x64 Icons
Ign:30 http://ppa.launchpad.net/fcl-debs/ppa/ubuntu xenial/main amd64 Packages 
Get:49 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main amd64 DEP-11 Metadata [304 kB]
Ign:31 http://ppa.launchpad.net/fcl-debs/ppa/ubuntu xenial/main i386 Packages  
Get:50 http://cn.archive.ubuntu.com/ubuntu xenial-updates/main DEP-11 64x64 Icons [203 kB]
Ign:32 http://ppa.launchpad.net/fcl-debs/ppa/ubuntu xenial/main all Packages   
Ign:33 http://ppa.launchpad.net/fcl-debs/ppa/ubuntu xenial/main Translation-en_US
Get:51 http://cn.archive.ubuntu.com/ubuntu xenial-updates/universe amd64 Packages [510 kB]
Ign:36 http://ppa.launchpad.net/fcl-debs/ppa/ubuntu xenial/main Translation-en 
Ign:38 http://ppa.launchpad.net/fcl-debs/ppa/ubuntu xenial/main amd64 DEP-11 Metadata
[... cut ...]
W: Target Translations (robotpkg/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list.d/robotpkg.list:2 and /etc/apt/sources.list.d/robotpkg.list:3
W: Target DEP-11 (robotpkg/dep11/Components-amd64.yml) is configured multiple times in /etc/apt/sources.list.d/robotpkg.list:2 and /etc/apt/sources.list.d/robotpkg.list:3
W: Target DEP-11-icons (robotpkg/dep11/icons-64x64.tar) is configured multiple times in /etc/apt/sources.list.d/robotpkg.list:2 and /etc/apt/sources.list.d/robotpkg.list:3
```

5.So ```/etc/apt/sources.list.d/robotpkg-sot.list``` gives me ```bash: /etc/apt/sources.list.d/robotpkg-sot.list: No such file or directory```

jcarpent, 2017-08-21:
@longwoo Do you still have some troubles when installing Pinocchio?

longwoo, 2017-08-21:
@jcarpent I have successfully installed from source last time.  But still in trouble with cmd ```sudo apt-get install robotpkg-pinocchio``` now, it gives me the same message as before.  Maybe I have to remove it first.

closed 2017-10-02 by jcarpent