stack-of-tasks/pinocchio #397: Move radius info from GeometryData to GeometryModel

jcarpent, 2017-08-10:
radius quantity is only depending on the GeometryModel and it is not varying with time.
@florent-lamiraux, @jmirabel I would suggest to shift radius to GeometryModel, and then update radius when adding a new geometry to the model.

jmirabel, 2017-08-21:
The impact should be rather small but not neglectible as it is used in a
few places.

However, there are two different things: the radius of each body and the
radius of a joint. The radius used in HPP is the associated to joints, i.e.
it encapsulates all the geometries attached to a joint. So it depends on
Model and GeometryModel.

I did not double-check but I think:
- the radius of each geometry model can be found in the fcl object itself;
- the radius in GeometryData correspond to joint radius.

nmansard, 2017-08-21:
I confirm that the radius in data is the result of a post-process of 
model, hence is better located in data.

jcarpent, 2017-08-21:
@jmirabel I was indeed mentioning the radius of a joint.

@ nmansard But they are quantities that are just fixed as soon as GeomModel is given. I do not think good to propagate such information inside GeomData, while these quantities are the same for all GeomData.

jmirabel, 2017-08-21:
> they are quantities that are just fixed as soon as GeomModel is given.

As soon as Model and GeomModel are given.

jcarpent, 2017-08-21:
Indeed, they can be computed by insertion.

jcarpent, 2018-06-08:
I closed this discussion because this issue is no more essential.

closed 2018-06-08 by jcarpent