stack-of-tasks/pinocchio #398: [python] Warning at quaternion multiplication.

aelkhour, 2017-08-14:
- pinocchio v1.2.3
- eigenpy 1.3.1

This might only be related to eigenpy: when I run the following python script 

```python
from pinocchio import Quaternion

q1 = Quaternion(1, 2, 3, 4)
q2 = Quaternion(5, 6, 7, 8)
q1 * q2
```

I systematically get the following warning in the console output:
```shell
The python object is not a numpy array.
```

I would like to know whether there is any known workaround to avoid printing this warning?

jcarpent, 2017-08-14:
Hi Antonio,

I do not have this warning on OS X with last devel version of Pinocchio + EigenPy 1.3.2
Can you check with the last version of EigenPy? I guess it should solve your problem.

aelkhour, 2017-08-14:
Indeed I ran the same test with eigenpy 1.3.2, and it seems to have solved the issue, thanks!

closed 2017-08-14 by aelkhour