stack-of-tasks/pinocchio #417: Use of the frame jacobian

MaximilienNaveau, 2017-11-28:
Dear all,

I found a unsettling result in Pinocchio, and I am not sure if it is a bug or my misunderstanding of the kinematics.
I am extensively using the operational frames in order to do inverse kinematics, forward kinematics and so forth.

In the following script I am trying to understand how the frame_jacobian is related to the joint_jacobian.
I am basically testing the following.
frame_M_joint * jac_joint_local == jac_frame_local
world_M_joint * jac_joint_local == jac_joint_global
world_M_frame * jac_frame_local == jac_frame_global

[sandbox2.py.zip](https://github.com/stack-of-tasks/pinocchio/files/1510825/sandbox2.py.zip)

The frist two test are true, though not the last one.
Why is that? I am sure there is something wrong in the last test but I can't figure out what.

Best

jcarpent, 2017-11-28:
@MaximilienNaveau Thanks for your issue.
In fact, both functions return different results in terms of interpretation:

- getJacobian<false> returns the Jacobian of the **center of the JOINT** expressed in the JOINT frame
WHILE getJacobian<true> returns the Jacobian of the **point coincident with center of the _WORLD_ but attached to the JOINT frame** and this Jacobian expressed in the WORLD frame

- getFrameJacobian<false> returns the Jacobian of the **center of the FRAME** expressed in the  FRAME frame 
WHILE getJacobian<true> returns the Jacobian of the **center of the _FRAME_ still** and expressed in the WORLD frame

In fact, we must modify this convention to always return the Jacobian of the same FRAME (e.g. center of the JOINT) but expressed either in the local frame of the JOINT or in the WORLD frame

MaximilienNaveau, 2017-11-28:
Thank you very much Justin for the clarification, I fixed my code accordingly.
I implemented a RobotWrapper to deal with this sort of things.

jcarpent, 2017-11-30:
Expecting everything goes well now.

closed 2017-11-30 by jcarpent