stack-of-tasks/pinocchio #418: Uniformization of the output of getJacobian and getFrameJacobian

jcarpent, 2017-11-28:
Following the discussion #417, I think it is necessary to uniform the output of getJacobian and getFrameJacobian.

I suggest that both of them returns the Jacobian of the center of the {JOINT,FRAME} either expressed in the LOCAL or WORLD frame.

@nmansard, @jmirabel, @andreadelprete What do you think about this modification as it impacts mainly your codes?

florent-lamiraux, 2017-11-28:
I would suggest to keep expressing velocities in the local frame since we modify our code to follow this convention.

jcarpent, 2017-11-28:
Certainly I was not clear enough. 
The problem is that currently getJacobian<local> returns 

- the Jacobian of the coincident point with the center of the WORLD (but attached to the JOINT) if local is set to True
- the Jacobian of the center of the JOINT if local is set to False

In other words, depending on the template parameters, the matrix returned by getJacobian<local> does not refer to the same geometrical object.

MaximilienNaveau, 2017-11-28:
> Following the discussion #417, I think it is necessary to uniform the output of getJacobian and getFrameJacobian.

I think this is a really good idea

jmirabel, 2017-11-29:
Is the fact that `getJacobian` and `getFrameJacobian` are template functions used somewhere ?
In the API, I think the template is counterproductive. It is very easy to forget whether `getJacobian<true>` means local or global frame.

I agree with @jcarpent's suggestion.

olivier-stasse, 2017-11-29:
Instead of a boolean ze could use an enum (seen as a type) explicitely refering to the convention. If no default value is provided, an error at compile time would force the user to specify the frame. We could have a look to eigen to generate a comprehensive compile error message if the case occur.

nmansard, 2017-11-29:
API is a consequence of how the jacobian are computed. 

1.In getJacobian(), we first compute a dense matrix containing the spatial "rotation" axis  of each joint in world frame (using computeJacobians() ), then copy the necessary columns (using getJacobian() ). The consequence is that getJacobian() should nominally returns the world jacobian(). We added for convenience an extra parameter to change the frame (from world to local) with extra computation. From the nature of the algo, the extra cost is minor.

2.In "jacobian()" we compute only one jacobian by back propagation along the kine tree from the end-eff of interest. The consequence is that jacobian() returns a jacobian in the local frame.

3.There is no sense in having getFrameJacobian() in world frame, as, in the world frame, joint jacobian and frame jacobian are equal.

Possible actions: in any case, I think we should keep the world option, as it is the most efficient.

1. We can remove the template boolean, but with extra cost. Maybe not important. Possibly it would be more productive to add alias function getJacobianWorld() and getJacobianLocal(). We can also alias true and false with LOCAL and WORLD.

2. We can add a boolean parameter in jacobian() to have the option of world expression. Really useful? In any case, computeJacobians() + getJacobian() should be preferred.

3. Adding a world option here has no meaning.

olivier-stasse, 2017-11-29:
IMHO The python interface would be clearer by explicitly calling the Kinematic update and with getJacobianWorld and getJacobianLocal.

andreadelprete, 2017-11-29:
My only request would be to wait until mid December before doing any change that could impact the library TSID. We have limited time to finish our experiments on the robot, and it would be a pity to waste time debugging issues introduced by this change.

jcarpent, 2017-12-12:
Solved by #418.

closed 2017-12-12 by jcarpent