stack-of-tasks/pinocchio #423: Cylinders in URDF are parsed as Capsule

jmirabel, 2017-12-05:
When encountering tag cylinder in URDF, pinocchio parser builds a `fcl::Capsule`. The reason for this is that URDF does not support capsules (and it does not seem it will...).

I see several way to handle this:
1. use a negative radius and or length to specify that a cylinder should be a capsule,
2. add an attribute or a tag below tag `cylinder` in the URDF and parse it with pinocchio,
3. use the tag `collision_checking` mentioned at the bottom of [this page](http://wiki.ros.org/urdf/XML/link#Multiple_Collision_Bodies),
3. do not support capsule.

I can work on this. Any preference ?

jcarpent, 2017-12-05:
I think 
1. is the shortest to handle your cases
BUT
2. might be investigated as a long term solution as we will be able to handle new features that are not provided by URDF yet.

jcarpent, 2017-12-05:
In conclusion, I would go for the second solution.

olivier-stasse, 2017-12-05:
As the capsule are mostly used for doing collision checking and that this is ignored by the URDF parser why @jcarpent do you prefer 2 over 3 ?

jcarpent, 2017-12-05:
Just be because in the two scenarios, we will need to add an additional parsing stage. So you're right, 2. and 3. are equivalent, except we can give precise naming with the second option.

olivier-stasse, 2017-12-05:
yes but if we use 3 and if the block checking_collision is ignored we could use directly capsule in place of cylinder which to me looks more readable.

jmirabel, 2017-12-05:
According to the discussion, I chose solution 3.

jcarpent, 2017-12-05:
Solved by #425.

PR #425 (2017-12-05): Handle capsules and cylinders in a cleaner way.
Fixes #423
closed 2017-12-05 by jcarpent