stack-of-tasks/pinocchio #427: Problem with mprint function

zhangOSK, 2017-12-29:
Hello there!
I am a new user for pinocchio. When I follow Tutorails 1.0, there are some errors. Could you please help tell me how to solve them?
[image]
[image]

Another thing is currently the latest versions of omniORB is omniORB 4.2.2 and omniORBpy 4.2.2. But for me, it is not working. Perhaps it's better to mention in the installation part that the old version (ORB4.1.x and ORBpy 3.x) is preferred. 
Thanks!

jcarpent, 2018-01-02:
Hi,
We forgot to overload the operator* for vector of dimension 3 types. I will fix it soon.
Please use `op = oM1.act(_1p)` instead.

zhangOSK, 2018-01-02:
Thanks for your reply and happy new year!
The other problem I met is 'Model' object has no attribute 'addJointAndBody'.
Could you please also help me solve this problem. Thank you a lot.

[image]

jcarpent, 2018-01-02:
Happy new year too!

To replace this method `addJointAndBody`, you can first call `addJoint` followed by `appendBodyToJoint`. It will do the same.

jcarpent, 2018-01-02:
I've just fixed the first issue with commit 98481df18bf7ea7a8ef070306adee3521b8bbb9c. 
You can update your repository and proceed to a new installation.

zhangOSK, 2018-01-02:
Thanks for your help!
I tried to use addJoint, but I met a augment error.  
What does the "Model" or "se3::Model{lvalue}" mean?

[image]

Besides, I used robotpkg to installed pinocchio, how can I make change of the first issue by robotpkg?

jcarpent, 2018-01-02:
`addJoint` takes four arguments:
- parent
- jointModel
- jointPlacement
- jointName
and `appendBodyToJoint` takes three arguments:
- parent_id (id of the supporting joint)
- bodyInertia
- bodyPlacement

And `se3::Model{lvalue}`is equivalent of the Python self argument for methods. It is automatically generated by boost-python.

I will update soon pinocchio on robotpkg to fix your issues.
Best,

Justin

zhangOSK, 2018-01-02:
Thanks a lot! 
Now the addJoint and appendBodyToJoint function work well.
However,, another issue appeard
[image]
How can I fix this,

zhangOSK, 2018-01-02:
And also missing of geometry, like this
[image]

jcarpent, 2018-01-02:
I guess you are on very old tutorials. @nmansard Can you provide the most recent versions of your tutorials?

There is no more model.nbody parameter. You can remove this info from printing.
`se3.geometry` has been replaced by `se3.forwardKinematics`.

nmansard, 2018-01-07:
See https://github.com/stack-of-tasks/pinocchio-tutorials

jcarpent, 2018-01-10:
@zhangOSK Did you fix your issues?

zhangOSK, 2018-01-10:
Thank you!
I have fixed the issue with your help!

jcarpent, 2018-01-10:
Perfect!

closed 2018-01-10 by jcarpent