stack-of-tasks/pinocchio #430: spatial velocity computed by forwardKinematics?

zongmianli, 2018-01-12:
After a function call of se3::forwardKinematics(model, data, q, vq), are the spatial velocities stored in data.v[j] expressed in world frame (i.e., ^o\nu_{o,j}) or the joint's local frame (^j\nu_{o,j})? Originally and after checking the source code I believe it's the later. Could you confirm this?

jcarpent, 2018-01-12:
Local frame indeed. Please refer to Featherstone's book for such details.

closed 2018-01-12 by jcarpent

zongmianli, 2018-01-12:
Thanks @jcarpent !

jcarpent, 2018-01-12:
@zongmianli Your welcome.

nmansard, 2018-01-12:
Featherstone is a good reference for the general organization. Unfortunately, the details of the algorithms is not the same, so some times it is a good idea to look in the textbook, some times it is not. 

@zongmianli , if you have an idea where to put this information in the code comments, could you make a PR with it?