stack-of-tasks/pinocchio #432: Uniformization of function naming

jcarpent, 2018-01-17:
I would like to change getJacobian and computeJacobians to getJointJacobian and computeJointJacobians.
The idea is to make uniform the naming of functions related to Joint and Frame Jacobians.

What do you think?

olivier-stasse, 2018-01-17:
It will be more accurate.

proyan, 2018-01-17:
For the sake of uniformization, I am also in favor of using the template reference frames LOCAL/WORLD for getFrameJacobian. getFrameJacobian<WORLD> could just be a wrapper that calls getJointJacobian of the parent. But this would atleast complete the API, and make the distinction clearer.

jcarpent, 2018-01-17:
We already discussed this precise point during a past HPP meeting.

jmirabel, 2018-02-03:
Before doing such an API change, if it hasn't been done yet, I propose to list all the names that would benefit from a change. I have in mind:

- [ ] `computeJacobians` -> `computeJointJacobians`
- [ ] `getJacobians` -> `getJointJacobians`
- [ ] `LieGroupTpl` -> `LieGroupMap` because it maps joint types to Lie group types, i.e. `LieGroupTpl::operation<JointModelType>::type` is a class deriving from `LieGroupOperationBase`.

jcarpent, 2018-02-03:
It is indeed not done yet.
All the naming change seems valid and coherent with the standardization of the API.

As @proyan mentioned, we may also investigate the standardization of templates for both `getJointJacobian` and `getFrameJacobian`.

And also for `getFrameJacobian`, I think we can avoid a call to `forwardFrameKinematics` by just updating the frame placement in the corpus of `getFrameJacobian`. This is just a suggestion to avoid additional computations which are not required.

If everyone agrees, I will make this change next week.

jmirabel, 2018-02-06:
I would also remove the `Operation` from all the `LieGroupOperationBase` derived classes. The class names are long and `Operation` does not add any pieces of information.

jmirabel, 2018-02-06:
Also, the terms `difference` and `differentiate` are both used for the same meaning. I suggest to only use one of them. (IMHO, `difference` is better as the operation that are done are not limited to small displacements)

closed 2018-07-11 by jcarpent