stack-of-tasks/pinocchio #437: Computing gradients and hessians

Bilalhmd, 2018-02-09:
Is there an implementation to compute gradients and hessians of the dynamics ? i can find a finite-difference implementation in algorithms, will I be able to use this ?

jcarpent, 2018-02-09:
Which derivatives do you need?

Bilalhmd, 2018-02-09:
rearranged my dynamics into the form dx/dt = f(x,u), wondering if there is functions to compute df/dx, df/du etc .. or if there is a general way that these are computed when solving typical optimization problems 

Thanks alot

cmastalli, 2018-02-09:
I would take this opportunity to open a discussion about automatic differentiation. I know that we can find analytical derivatives of some rigid body algorithms, but do you think we can do it for all the algorithms? In negative case, it would be nice to extend Pinocchio with CppAd

jcarpent, 2018-02-09:
@Bilalhmd  It really depends on the choice of your robot. Is a free floating base system or not.
I would really recommand you to first use finite differences for prototyping your codes.
We already provide some derivatives for doing such kind of computations. 
We hope to be able to integrate the analytical derivates in Pinocchio, but it remains a work in progress.

jcarpent, 2018-02-09:
@cmastalli It would be interested indeed, but AD does not solve all the problems. It is also a hard to make all the code compliant with operator overloading.
We let this task as a very low priority.

jcarpent, 2018-02-20:
@Bilalhmd Do you need additional help for derivatives? I will close this issue.
Feel free to open it again if you have other questions.

closed 2018-02-20 by jcarpent

jcarpent, 2018-07-10:
@Bilalhmd A while ago, you asked for computing derivatives in Pinocchio.
This message just to inform you that they are now available on the devel branch of Pinocchio.
There is a RSS paper which explains the computations: https://hal.laas.fr/hal-01790971

Best,

Justin

hammoudbilal, 2018-11-06:
Hi, this have been really helpful, it seems to me that the terms regarding the contact/external forces are not included in the code, for example in the forward dynamics derivative  dFD/dq should i just simply add (dM_inv/dq).J^T.f_{ext}  + M_inv.(dJ^T/dq).f_{ext}  ?

best 
Bilal

jcarpent, 2018-11-06:
They are taken into account on the devel branch.

hammoudbilal, 2018-11-06:
found it, thanks! it seems that only the python binding was missing for the extended version of the function, i will stop posting on this thread.

jcarpent, 2018-11-06:
Do you need them?

hammoudbilal, 2018-11-07:
no everything works fine, thanks again