stack-of-tasks/pinocchio #440: Actuation, transmission and mimic joints

jcarpent, 2018-02-22:
@olivier-stasse together with @flforget have recently opened a PR #434 concerning the adding of DC motor models into Pinocchio in order to to tack also into account the dynamical properties of these components into rigid body dynamics algorithms (inertia effects, etc).

To be effective, we also need to consider the transmission parts, namely gear reducers, pulley-belt systems. It is then important to notice that such transmission model can also handle mimic joints, like the joints that appear in the gripper of HRP-2 or Pyrene humanoids.

The idea of this issue is to collect all the concerns about adding actuation and transmission into Pinocchio.

aelkhour, 2018-02-22:
It can be useful to handle complex transmissions such as differential drives, where N joints are actuated by M motors.

This seems to be well handled by the ros transmission_interface API.

jcarpent, 2018-03-07:
#442 is a temporary inclusion of a simple transmission and motor inertia parameters.
This will be extended through #434.