stack-of-tasks/pinocchio #445: How does pinocchio parse floating joint?

longwoo, 2018-03-19:
In URDF , I have a floating joint ,how does pinocchio parse it? For example ,the `NQ` and `NV` is 7 and 6 in a floating joint. If I want to give a init q to the floating joint. I should give an array like this ```init_q=(1,2,3,4,5,6,7)```. The question is in which order? If `X,Y,Z` is translation，`x,y,z,w` is rotation. The order should be ```(X,Y,Z,x,y,z,w)``` or other form?

jmirabel, 2018-03-19:
Your guess is correct: `(X,Y,Z,x,y,z,w)`

jcarpent, 2018-03-19:
Are your working under Python? If so, you can use the se3ToXYZQUAT function with transforms a SE3 placement into the right convention for the freeflyer.

longwoo, 2018-03-19:
@jmirabel OK, Thank you.

longwoo, 2018-03-19:
@jcarpent Yes , I am using Python. The motivation for this is the usage of `RobotWrapper.Jcom`,`RobotWrapper.mass` and `RobotWrapper.bias`,so I need a list of q and v.

jcarpent, 2018-03-19:
Can you report on the mismatches if any?
Thanks in advance

jcarpent, 2018-03-20:
@longwoo Feel free to open the issue if needed.

closed 2018-03-20 by jcarpent

cmastalli, 2018-03-20:
@longwoo  I would like to add a comment. Pinnocchio is aligned with Gazebo standard. As you might know, Gazebo doesn't parse floating-base joints. If it's possible for you, you should define the floating-base joint as fixed one in your URDF, and then declaring it as FreeFlyer joint in the RobotWrapper (c++ or Python code). Here you have an example in Python:

import pinnochio as pin
pin.robot_wrapper.RobotWrapper(urdf, [path,], pin.JointModelFreeFlyer())

longwoo, 2018-03-21:
@cmastalli Thank you the comment. AS a test urdf, I defined it as 
```
<?xml version="1.0" ?>
<robot name="floating_robot" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <link name="base_link"/>
  <joint name="v_joint" type="floating">
    <origin rpy="0 0 0" xyz="0 0 0"/>
    <parent link="base_link"/>
    <child link="load"/>
    <axis xyz="0 0 0"/>
  </joint>
  <link name="load">
    <inertial>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <mass value="35"/>
      <inertia ixx="0.0583" ixy="0" ixz="0" iyy="0.0583" iyz="0" izz="0.0583"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <box size="0.1 0.1 0.1"/>
      </geometry>
      <material name="">
        <color rgba="0.866666666666667 0.909803921568627 1 1"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <box size="0.1 0.1 0.1"/>
      </geometry>
    </collision>
  </link>
</robot>
```
But I didn't add a `pin.JointModelFreeFlyer()` as init ,thanks for mention.
The point is if I add the `pin.JointModelFreeFlyer()` as init, the `model.nv' and  'model.nq` value become 12 and 14. Is that means if I add a floating joint in urdf then I don't have to add the `pin.JointModelFreeFlyer()` as init? Or if I add the `pin.JointModelFreeFlyer()` then I don't have to add  floating joint in urdf ? They are equal?
By the way my pinocchio is installed from source last year.  I am trying to up to date.