stack-of-tasks/pinocchio #446: floating joint Problem in Jcom

longwoo, 2018-03-19:
My urdf is 
```
<?xml version="1.0" ?>
<robot name="floating_robot" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <link name="base_link"/>
  <joint name="v_joint" type="floating">
    <origin rpy="0 0 0" xyz="0 0 0"/>
    <parent link="base_link"/>
    <child link="load"/>
    <axis xyz="0 0 0"/>
  </joint>
  <link name="load">
    <inertial>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <mass value="35"/>
      <inertia ixx="0.0583" ixy="0" ixz="0" iyy="0.0583" iyz="0" izz="0.0583"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <box size="0.1 0.1 0.1"/>
      </geometry>
      <material name="">
        <color rgba="0.866666666666667 0.909803921568627 1 1"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <box size="0.1 0.1 0.1"/>
      </geometry>
    </collision>
  </link>
</robot>
```
My test code is 
```
import pinocchio as pin
from pinocchio.robot_wrapper import RobotWrapper
import numpy as np

## Load Romeo with RomeoWrapper
import os
current_path = os.getcwd()

# The model of Romeo is contained in the path PINOCCHIO_GIT_REPOSITORY/models/romeo
model_path = current_path + "/" + "../../models/floating_test"
mesh_dir = model_path
urdf_filename = "floating.urdf"
urdf_model_path = model_path + "/urdf/" + urdf_filename
robot = RobotWrapper(urdf_model_path, [mesh_dir])
model = robot.model
data = robot.data
rq=np.asarray([-0.0410174, -0.00164758, 0.70415, -7.37661e-06, -0.00920039, -0.00565861, 0.999942])
rv=np.asarray([-0.00678374, 0.00192725, -0.00186328, -0.00307152, -0.0140679, -0.000485445])
com2 = pin.centerOfMass(model,data,rq)
print "com2", com2
print "model.nv", model.nv
print "model.nq", model.nq
Jc = pin.se3.jacobianCenterOfMass(robot.model,robot.data,rq)
Jc1 = robot.Jcom(rq)
print "Jcom=", Jc
print "Jc1==========", Jc1
print "mass=", robot.mass(rq)
print "bias=", robot.bias(rq, rv)#-robot.gravity(rq)

```
The result is 
```
Jcom=
 [[  9.99766666e-01   1.13166993e-02  -1.83996293e-02   1.58603289e-18
    1.01506105e-16  -1.98254112e-19]
 [ -1.13164279e-02   9.99935960e-01   1.18875202e-04   0.00000000e+00
    0.00000000e+00   0.00000000e+00]
 [  1.83997962e-02   8.93704734e-05   9.99830706e-01   0.00000000e+00
    0.00000000e+00   0.00000000e+00]]
```

My question is Why the Jcom is Not something like
```
 [[  1   0 0   1.58603289e-18    1.01506105e-16  -1.98254112e-19]
  [ 0   1   0  0.00000000e+00    0.00000000e+00   0.00000000e+00]
  [  0   0   1   0.00000000e+00    0.00000000e+00   0.00000000e+00]]
```
Jcom[0,1] and Jcom[0,2] are so much bigger than 0  that I can't not treat them as calculate error. Intuitivly, the Jacobian of a transition term should be identity matrix. Did I miss something or make a mistake somewhere?

jcarpent, 2018-03-19:
It is just a matter of frames. In Pinocchio, the Jacobian of center of mass is expressed in the world frame. While in Mujoco, it seems to be expressed in the local frame of the waist.

To check that, you can just retrieve the Jacobian of the CoM expressed in the local frame through:
`data.oMi[1].rotation.T * Jc`

longwoo, 2018-03-20:
@jcarpent Thank you for your help. After add the rotation you mentioned. The error between Pinocchio and Mujoco reduced a lot in my test leg urdf with floating base. Here is the Jcom, Inertia Matrix (H) and bias force (C) result in Mujoco at state:
```
rq=np.asarray([-0.0410174, -0.00164758, 0.70415, -7.37661e-06, -0.00920039, -0.00565861, 0.999942, 3.35967e-05, -0.299547,
               -3.68394e-06, 0.80442, -0.49913, -1.61726e-05, 2.36842e-06, -0.299507, 7.08779e-06, 0.804448, -0.499125, -2.10858e-05])
rv=np.asarray([-0.00678374, 0.00192725, -0.00186328, -0.00307152, -0.0140679, -0.000485445, 1.18537e-05, 0.00599099, 2.96269e-05,
               0.0136278, 0.00270404, 0.00172085, 0.000326742, 0.00669952, 1.02618e-05, 0.0118339, 0.00248553, 0.00121971 ])
```
```
mujoco_Jc = np.asarray([[1,0,0,-0.00035697,0.0342082,0.00164355,0.000678503,-0.0594422,0.000132229,-0.0182478,-0.0001486,2.31793e-06,0.000695408,-0.0594424,0.000557303,-0.0182474,-0.000148928,2.31596e-06],
                        [0,1,0,-0.0337743,-0.000384976,0.023901,0.0595863,0.00067372,-0.0106049,0.000205907,2.24618e-06,0.000204764,0.0595863,0.000673492,-0.0106102,0.000205906,2.0936e-06,0.000204764],
                        [0,0,1,-0.0013761,-0.023293,2.74582e-05,-0.000221377,-0.00838582,8.16709e-05,0.00514066,-0.00443508,2.23782e-08,-0.00114059,-0.00838294,0.000221883,0.00514192,-0.00443507,1.43258e-08]])
```
```
mujoco_H = np.asarray([
    [ 69.645,            0,            0,   -0.0248612,      2.38243,     0.114465,    0.0472544,     -4.13985,   0.00920907,     -1.27087,   -0.0103492,  0.000161432,    0.0484317,     -4.13986,    0.0388134,     -1.27084,   -0.0103721,  0.000161295],
    [      0,       69.645,            0,     -2.35221,   -0.0268117,      1.66459,      4.14989,    0.0469212,    -0.738577,    0.0143404,  0.000156435,    0.0142608,      4.14989,    0.0469053,    -0.738945,    0.0143403,  0.000145809,    0.0142608],
    [      0,            0,       69.645,   -0.0958384,     -1.62224,   0.00191233,   -0.0154178,     -0.58403,   0.00568797,     0.358022,    -0.308881,  1.55853e-06,   -0.0794361,     -0.58383,     0.015453,     0.358109,     -0.30888,  9.97719e-07],
 [-0.0248612,     -2.35221,   -0.0958384,     7.88794,   -0.00218478,    -0.122458,      2.19823,   -0.0369751,    -0.465197,     0.024069,   -0.0227576,    0.0114826,      2.20493,    0.0342496,     -0.46662,   -0.0325146,    0.0222577,    0.0114826],
   [  2.38243,   -0.0268117,     -1.62224,  -0.00218478,      7.98359,   -0.0334353,   -0.0002794,     2.25977,  -0.00595064,     0.816561,    0.0415288, -6.36294e-08,  -0.00189137,      2.25975,    -0.024931,     0.816554,    0.0415328,  1.11158e-07],
  [  0.114465,      1.66459,   0.00191233,    -0.122458,   -0.0334353,     0.724589,     0.190284,     0.283741,    0.0140359,    0.0858676,   0.00112996,   0.00143335,     0.190178,    -0.317173,    0.0186179,    -0.100177,  -0.00112732,   0.00143269],
[   0.0472544,      4.14989,   -0.0154178,      2.19823,   -0.0002794,     0.190284,      2.19934,  -0.00142819,    -0.465577,  -0.00262396,   -0.0011529,    0.0114825,            0,            0,            0,            0,            0,            0],
 [   -4.13985,    0.0469212,     -0.58403,   -0.0369751,      2.25977,     0.283741,  -0.00142819,      2.25977,  -0.00595012,     0.816561,    0.0415288, -1.54903e-08,            0,            0,            0,            0,            0,            0],
[  0.00920907,    -0.738577,   0.00568797,    -0.465197,  -0.00595064,    0.0140359,    -0.465577,  -0.00595012,     0.151982,    -0.001758,  0.000346553,   -0.0020189,            0,            0,            0,            0,            0,            0],
 [   -1.27087,    0.0143404,     0.358022,     0.024069,     0.816561,    0.0858676,  -0.00262396,     0.816561,    -0.001758,     0.420052,   0.00301517, -9.42349e-10,            0,            0,            0,            0,            0,            0],
[  -0.0103492,  0.000156435,    -0.308881,   -0.0227576,    0.0415288,   0.00112996,   -0.0011529,    0.0415288,  0.000346553,   0.00301517,    0.0477842,  3.94896e-08,            0,            0,            0,            0,            0,            0],
[ 0.000161432,    0.0142608,  1.55853e-06,    0.0114826, -6.36294e-08,   0.00143335,    0.0114825, -1.54903e-08,   -0.0020189, -9.42349e-10,  3.94896e-08,   0.00240541,            0,            0,            0,            0,            0,            0],
[   0.0484317,      4.14989,   -0.0794361,      2.20493,  -0.00189137,     0.190178,            0,            0,            0,            0,            0,            0,      2.19934,  -0.00128323,    -0.465595,   -0.0058155,  0.000653091,    0.0114826],
 [   -4.13986,    0.0469053,     -0.58383,    0.0342496,      2.25975,    -0.317173,            0,            0,            0,            0,            0,            0,  -0.00128323,      2.25975,   -0.0249309,     0.816554,    0.0415328,  1.14083e-07],
 [  0.0388134,    -0.738945,     0.015453,     -0.46662,    -0.024931,    0.0186179,            0,            0,            0,            0,            0,            0,    -0.465595,   -0.0249309,     0.151992,  -0.00945394, -0.000196407,  -0.00201909],
 [   -1.27084,    0.0143403,     0.358109,   -0.0325146,     0.816554,    -0.100177,            0,            0,            0,            0,            0,            0,   -0.0058155,     0.816554,  -0.00945394,     0.420053,   0.00301564, -1.17426e-09],
[... cut ...]
 [ -1.12093164e-04]
 [ -1.96075912e-04]
 [  2.46485668e-05]
 [ -1.61601876e-05]
 [  3.64742698e-05]
 [ -2.44007619e-07]]
```

As we can see that the Jcom is much closer now , but some elements in the Joint Space Inertia Matrix and Joint Space Bias Force  still much different.

By the way, where can I find the complete source code example using pinocchio to control the Romeo or Simple Humanoid Model ZMP or Inverse Dynamic walking? There are limit  info in http://stack-of-tasks.github.io/pinocchio/tutorials.html.

jcarpent, 2018-03-20:
I'm very afraid by the differences between the two frameworks.
For pinocchio, it is validated against RBDL and Metapod.
Can you send me a complete zip with all your code and models in order I can check? (by email if you prefer).

For walking examples, you can use the tutorial made by Nicolas: https://github.com/stack-of-tasks/pinocchio-tutorials with the text in here.

longwoo, 2018-03-21:
@jcarpent Thank you for your attention. I have sent you an email at  jcarpent@laas.fr . Please let me know if the code not work.

jcarpent, 2018-03-22:
Thank you for your feedback. I close the issue, do not hesitate to reopen it.

closed 2018-03-22 by jcarpent

longwoo, 2018-04-03:
@jcarpent May be the different is according to [http://www.master-ris.unina.it/custom-php/materiale_didattico/bruno_siciliano_1300033341.pdf](url) 
There are two definition of Jacobian, Geometric Jacobian (Jg) and Analytical Jacobian (Ja).
One is mapping to angular angles while the other mapping to Euler angles.
So I guess in pinocchio, it means Geometric Jacobian.

jcarpent, 2018-04-03:
@longwoo In Pinocchio, the Jacobian corresponds to the variation of the placement variable according to a variation of the joint configuration. So it maps from the tangent space of the configuration space (denoted v in Pinocchio) to the tangent of SE3 (denoted Motion in Pinocchio).

Hope it can help you understanding the concepts behind Pinocchio.
In the context of Jcom, it maps from the tangent of the configuration space to the velocity of the center of mass. This ones being expressed in the world frame in Pinocchio.

longwoo, 2018-04-11:
@jcarpent Thanks for your patient. Now ,there is no doubt that it is expressed in the world frame. One more question is in which point?
According to Feathstone's book Rigid Body Dynamics Algorithms , p10, linear velocity term ```V_o``` in spatial velocity (twist) means ```velocity of the body−fixed point that coincides with O at the current instant```. Suppose ```O``` is the origin of the world frame. Is that means, the result of ```v=J*dot_q``` is not the traditional physical meaning of v (for example ,the velocity of link pelvis frame origin w.r.t. the world frame), but the meaning of ```v_o```(for example ,the velocity of link pelvis−fixed point that coincides with O at the current instant w.r.t world frame)?
The motivation is that. If I want to solve ```dot_q``` ,I need to give a desired ```v```, so the first thing is to transform the ```v``` to the meaning of ```v_o```?

jcarpent, 2018-04-11:
The Jacobian can be expressed in two frames: WORLD or LOCAL:
For WORLD, the velocity indeed corresponds to the point coincident with the world origin, but linked to the joint and the twist is expressed in the world frame coordinates.
For LOCAL, it corresponds to the velocity of the center of the joint you consider expressed in the local coordinate frame of the joint.
Both Jacobians, J_in_world and J_in_local are linked through the spatial mapping:
J_in_world = action(data.oMi[k]) * J_in_local where k is the index of the joint you are considering, and action is the action of a placement quantity onto a twist.

Hopping it helps you!

longwoo, 2018-04-11:
@jcarpent Thank you. It helps a lot to find the different. And is it also the same to ```J_com```? The result of ```v_com=J_com*dot_q``` is corresponds to the point coincident with the world origin?

And I find a function in Mujoco 
```
// Compute object 6D velocity in object-centered frame, world/local orientation.
MJAPI void mj_objectVelocity(const mjModel* m, const mjData* d, 
                             int objtype, int objid, mjtNum* res, int flg_local);
```
which can get tradition meaning the velocity, not ```v_o```. Is there any similar function in pinocchio ? The way I use now is to do forward kinematic and then read the ```data.oMi[]```. Or a function that can transform ```v``` and ```v_o```?

jcarpent, 2018-04-11:
For the first question, vcom is the velocity of the center of mass  of the whole systems expressed in the world coordinate frame. This is a vector, and we chose to express in the world frame coordinate system. Nothing more.

Now, if you want to retrieve the velocity of a given joint defined by its id, you just need to call first `forwardKInemartics(model,data,q,qdot)`and then get the velocity of this joint by calling `data.v[id]`. This velocity will be expressed in the local coordinate system of the joint. Now, if you need to get this velocity expressed in the world frame, you just need to do:
`Motion v_local = data.v[id]; Motion v_global = Motion(data.oMi[id].rotation() * v_local.linear(), data.oMi[id].rotation() * v_local.angular());' which corresponds to rotate the linear and angular velocity vectors from one coordinate system to another one, without modifying the position of the point where the velocity is expressed.

@longwoo. Thank you a lot for these questions, they will help us to write our document for both the code and the convention that we use so far.

longwoo, 2018-04-12:
@jcarpent The different Inertia Matrix (H) between Mujoco and Pinocchio may caused by my misunderstanding about the input of freeflyer. I made 2 experiment .
First ,make the robot fixed in the air, then the robot have 12 degrees. In this case , they get the exactly same result like this
```
mujoco_q ====== 0.322425
-0.339424
 0.271903
 0.840166
-0.497537
  0.53392
 0.210648
-0.296649
  0.17112
 0.796374
-0.584511
0.0749731
robot->model().nq ======12
robot->model().nv ======12
pinocchio M      2.16345   0.00239819    -0.470753    -0.206435   -0.0134057    0.0100359            0            0            0            0            0            0
  0.00239819      2.22595    -0.124132     0.775033    0.0420481   0.00187402            0            0            0            0            0            0
   -0.470753    -0.124132     0.164512  0.000471574  0.000893136  -0.00204518            0            0            0            0            0            0
   -0.206435     0.775033  0.000471574     0.419111   0.00250398  2.34358e-05            0            0            0            0            0            0
  -0.0134057    0.0420481  0.000893136   0.00250398    0.0477037  -0.00112313            0            0            0            0            0            0
   0.0100359   0.00187402  -0.00204518  2.34358e-05  -0.00112313   0.00240541            0            0            0            0            0            0
           0            0            0            0            0            0        2.171  -0.00121148    -0.443089    -0.134955  -0.00340729    0.0111167
           0            0            0            0            0            0  -0.00121148       2.2235   -0.0968796     0.781072    0.0235894   0.00186572
           0            0            0            0            0            0    -0.443089   -0.0968796     0.147697  -0.00916737 -7.99017e-05  -0.00171924
           0            0            0            0            0            0    -0.134955     0.781072  -0.00916737     0.404667  -0.00467818  2.98019e-05
           0            0            0            0            0            0  -0.00340729    0.0235894 -7.99017e-05  -0.00467818    0.0477825 -0.000165305
           0            0            0            0            0            0    0.0111167   0.00186572  -0.00171924  2.98019e-05 -0.000165305   0.00240541
Mujoco M      2.16345   0.00239819    -0.470753    -0.206435   -0.0134057    0.0100359            0            0            0            0            0            0
  0.00239819      2.22595    -0.124132     0.775033    0.0420481   0.00187402            0            0            0            0            0            0
   -0.470753    -0.124132     0.164512  0.000471573  0.000893136  -0.00204518            0            0            0            0            0            0
   -0.206435     0.775033  0.000471573     0.419111   0.00250398  2.34358e-05            0            0            0            0            0            0
  -0.0134057    0.0420481  0.000893136   0.00250398    0.0477037  -0.00112313            0            0            0            0            0            0
   0.0100359   0.00187402  -0.00204518  2.34358e-05  -0.00112313   0.00240541            0            0            0            0            0            0
           0            0            0            0            0            0        2.171  -0.00121148    -0.443089    -0.134955  -0.00340729    0.0111167
           0            0            0            0            0            0  -0.00121148       2.2235   -0.0968796     0.781072    0.0235894   0.00186572
           0            0            0            0            0            0    -0.443089   -0.0968796     0.147697  -0.00916738 -7.99017e-05  -0.00171924
           0            0            0            0            0            0    -0.134955     0.781072  -0.00916738     0.404667  -0.00467818  2.98019e-05
           0            0            0            0            0            0  -0.00340729    0.0235894 -7.99017e-05  -0.00467818    0.0477825 -0.000165305
           0            0            0            0            0            0    0.0111167   0.00186572  -0.00171924  2.98019e-05 -0.000165305   0.00240541
```

Then I add the freeflyer joint  ,  and do a forward . Also get the  the exactly same result for a link pose
```
oMf   R =
  0.999512  0.0261185  -0.017138
 -0.025761   0.999453  0.0207621
 0.0176709 -0.0203105   0.999638
  p =  -0.0311666 -0.00697546    0.707958

pelvis_pos in mujoco  -0.0311666
-0.00697546
   0.707958
pelvis_mat in mujoco   0.999512  0.0261185  -0.017138
 -0.025761   0.999453  0.0207621
 0.0176709 -0.0203105   0.999638
```

The input ``q`` both get from mujoco , but the mismatch appears when freeflyer was added. And I get the ```q```  like this in freeflyer case
```
    Eigen::VectorXd pino_q = Eigen::Map<Eigen::ArrayXd>(&m_data->qpos[0], 19);//in mujoco floating base, qpos is (X,Y,Z,w,x,y,z)
    double t=0;//in pinocchio floating base, qpos is (X,Y,Z,x,y,z,w)
    t = pino_q(3);
    pino_q(3)=pino_q(4);
    pino_q(4)=pino_q(5);
    pino_q(5)=pino_q(6);
    pino_q(6)=t;
    pino_q.segment<4>(3) /= pino_q.segment<4>(3).norm(); // quaternion of freeflyer
```
So I think the reason cause the different in Inertia Matrix is freeflyer , but still have no idea what is wrong.

nmansard, 2018-04-12:
The inertia matrix represents a linear application from the tangent 
space of configurations (aka joint velocities or accelerations) to the 
cotangent space of configurations (aka joint torques):
tau_q = M a_q
With a free flyers, you would have something like tau_q = [ tau_ff; 
tau_act ] and a_q = [a_ff; a_act], with _ff the free flyer components 
and _act the components of the actuated joints.
In pinocchio, _ff are represented in the frame of the first body (both 
for motion a_ and forces tau_). Other choices are also acceptable. With 
another choice of representation for a_ and tau_, you would have a 
different representation of M (i.e different values in the six top first 
rows).

The M matrix can be easily validated against the RNEA algorithm. The 3 
first rows are also similar to the jacobian of the COM (in the proper 
frame -- in Pinocchio, J_com is in the world frame). As you likely have 
access to both RNEA and J_com in Mujoco, you may validate the computation.

longwoo, 2018-04-13:
Indeed, I made a test like this
```
    Eigen::VectorXd pino_q = Eigen::Map<Eigen::ArrayXd>(&m_data->qpos[0], 19);//in mujoco floating base, qpos is (X,Y,Z,w,x,y,z)
    Eigen::VectorXd pino_v = Eigen::Map<Eigen::ArrayXd>(&m_data->qvel[0], 18);
    pino_q(0)=0;
    pino_q(1)=0;
    pino_q(2)=0;
    pino_q(3)=0;
    pino_q(4)=0;
    pino_q(5)=0;
    pino_q(6)=1;
    tsid_robot->computeAllTerms(*tsid_data , pino_q, pino_v);
    std::cout <<"pinocchio M2\n "<<tsid_data->M << std::endl;

    mjModel * m_test_model;
    mjData * m_test_data;
    m_test_data = mj_makeData(m_model);
    mju_copy(m_test_data->qpos, m_data->qpos, m_model->nq);
    mju_copy(m_test_data->qvel, m_data->qvel, m_model->nv);
    mju_copy(m_test_data->qacc, m_data->qacc, m_model->nv);
    Eigen::MatrixXd m_M_test;
    m_M_test.resize(m_nv, m_nv);

    m_test_data->qpos[0]=0;
    m_test_data->qpos[1]=0;
    m_test_data->qpos[2]=0;
    m_test_data->qpos[3]=1;
    m_test_data->qpos[4]=0;
    m_test_data->qpos[5]=0;
    m_test_data->qpos[6]=0;
    mj_inverse(m_model,m_test_data);
    mj_fullM(m_model, m_M_test.data(), m_test_data->qM);
    std::cout <<"Mujoco M \n "<< m_M_test << std::endl;
    std::cout <<"Err M between Mujoco and Pinocchio \n "<< m_M_test - tsid_data->M << std::endl;
```
They get the same result when there is no rotation in freeflyer.
```
Err M between Mujoco and Pinocchio
            0            0            0            0 -4.41872e-10 -1.49741e-13            0  1.51763e-11  1.49979e-12 -5.95666e-11 -5.15596e-11 -2.20226e-13            0  1.07475e-10 -2.67186e-12 -1.14763e-10 -4.80575e-11 -2.18385e-13
           0            0            0  4.41872e-10            0 -1.47282e-12 -5.78115e-12  -9.8619e-16 -1.17945e-11 -2.66219e-13  -2.2738e-13  4.22751e-11 -1.02844e-10   4.0766e-15  2.61654e-11 -5.03061e-13 -2.13136e-13  4.08085e-11
           0            0            0  1.49741e-13  1.47282e-12            0  -3.8762e-13  4.66294e-14  1.00812e-13  2.22267e-13  4.06897e-14  4.23451e-15  5.37431e-13 -1.91236e-13 -5.94941e-14 -8.21565e-15  1.18794e-13  -1.4338e-13
           0  4.41872e-10  1.49741e-13  2.27374e-08  2.19771e-09 -4.53327e-08   1.1599e-08  6.15123e-10 -2.47322e-08 -1.19497e-09 -3.14471e-14   2.7658e-11  1.13885e-08  6.19638e-10 -2.47332e-08 -1.17628e-09 -3.85525e-14  2.69612e-11
-4.41872e-10            0  1.47282e-12  2.19771e-09 -2.62073e-10 -1.46254e-08  1.09417e-09 -1.16379e-10 -7.38181e-09 -4.36472e-10  3.37042e-11  1.43334e-13  1.10352e-09 -2.05669e-10  -7.3992e-09 -4.00425e-10  3.18714e-11  1.53067e-13
-1.49741e-13 -1.47282e-12            0 -4.53327e-08 -1.46254e-08 -2.30564e-08 -2.26353e-08 -7.57007e-09 -5.97385e-09 -2.04066e-09   4.6297e-12  2.73694e-12 -2.27005e-08 -7.57658e-09 -5.90626e-09 -2.06377e-09 -3.00018e-12  2.68269e-12
           0 -5.78115e-12  -3.8762e-13   1.1599e-08  1.09417e-09 -2.26353e-08   1.1599e-08  6.15102e-10 -2.47322e-08 -1.19496e-09  -5.3895e-14  2.76577e-11            0            0            0            0            0            0
 1.51763e-11  -9.8619e-16  4.66294e-14  6.15123e-10 -1.16379e-10 -7.57007e-09  6.15102e-10 -2.76464e-10 -7.50653e-09 -4.79629e-10  3.37183e-11  2.00874e-13            0            0            0            0            0            0
 1.49979e-12 -1.17945e-11  1.00812e-13 -2.47322e-08 -7.38181e-09 -5.97385e-09 -2.47322e-08 -7.50653e-09   -3.879e-11 -1.70995e-09  3.13178e-13 -3.64376e-12            0            0            0            0            0            0
-5.95666e-11 -2.66219e-13  2.22267e-13 -1.19497e-09 -4.36472e-10 -2.04066e-09 -1.19496e-09 -4.79629e-10 -1.70995e-09 -4.76052e-10  1.59009e-11 -5.26628e-15            0            0            0            0            0            0
-5.15596e-11  -2.2738e-13  4.06897e-14 -3.14471e-14  3.37042e-11   4.6297e-12  -5.3895e-14  3.37183e-11  3.13178e-13  1.59009e-11  1.18572e-12  1.27083e-13            0            0            0            0            0            0
-2.20226e-13  4.22751e-11  4.23451e-15   2.7658e-11  1.43334e-13  2.73694e-12  2.76577e-11  2.00874e-13 -3.64376e-12 -5.26628e-15  1.27083e-13  6.33534e-13            0            0            0            0            0            0
           0 -1.02844e-10  5.37431e-13  1.13885e-08  1.10352e-09 -2.27005e-08            0            0            0            0            0            0  1.13885e-08  6.19642e-10 -2.47332e-08  -1.1763e-09  -1.0636e-14  2.69512e-11
 1.07475e-10   4.0766e-15 -1.91236e-13  6.19638e-10 -2.05669e-10 -7.57658e-09            0            0            0            0            0            0  6.19642e-10 -3.67207e-10 -7.52335e-09 -4.44134e-10  3.18719e-11  2.10512e-13
-2.67186e-12  2.61654e-11 -5.94941e-14 -2.47332e-08  -7.3992e-09 -5.90626e-09            0            0            0            0            0            0 -2.47332e-08 -7.52335e-09 -3.49856e-11 -1.72968e-09 -3.06448e-13 -3.43874e-12
-1.14763e-10 -5.03061e-13 -8.21565e-15 -1.17628e-09 -4.00425e-10 -2.06377e-09            0            0            0            0            0            0  -1.1763e-09 -4.44134e-10 -1.72968e-09 -4.42398e-10  1.56685e-11  -3.3475e-15
-4.80575e-11 -2.13136e-13  1.18794e-13 -3.85525e-14  3.18714e-11 -3.00018e-12            0            0            0            0            0            0  -1.0636e-14  3.18719e-11 -3.06448e-13  1.56685e-11  1.11233e-12  1.45838e-13
-2.18385e-13  4.08085e-11  -1.4338e-13  2.69612e-11  1.53067e-13  2.68269e-12            0            0            0            0            0            0  2.69512e-11  2.10512e-13 -3.43874e-12  -3.3475e-15  1.45838e-13  6.11688e-13
```

So the inertia matrix in pinocchio will Not change w.r.t the input of first 7 element in ```q``` (the pos and orientation  of freeflyer w.r.t the world frame) by 
```
se3::crba(*robot_model,*tsid_data,pino_q);
```
However , In Mujoco,  the inertia matrix will change w.r.t the orientation  of freeflyer.
That is the different.