stack-of-tasks/pinocchio #448: How to solve the Pinocchio and Eigen conflict?

longwoo, 2018-03-22:
I have write a cmakelist.txt like this 
```
cmake_minimum_required(VERSION 2.8.3)
project(Test)

set(CMAKE_BUILD_TYPE Release)
set(CMAKE_CXX_FLAGS "-Wall -std=c++11 -mavx")
set(CMAKE_CXX_FLAGS_DEBUG "-g")
set(CMAKE_CXX_FLAGS_RELEASE "-O3")

include_directories(/opt/openrobots/include)
link_directories(/opt/openrobots/lib)

set(EIGEN3_INCLUDE_DIRS /usr/include/eigen3)
include_directories(${EIGEN3_INCLUDE_DIRS})

add_executable(Test main.cpp)
target_link_libraries(Test)
```

And a main.cpp like this
```
#include "pinocchio/spatial/fwd.hpp"
#include "pinocchio/spatial/se3.hpp"
#include "pinocchio/multibody/visitor.hpp"
#include "pinocchio/multibody/model.hpp"
#include "pinocchio/algorithm/aba.hpp"
#include "pinocchio/algorithm/rnea.hpp"
#include "pinocchio/algorithm/jacobian.hpp"
#include "pinocchio/algorithm/crba.hpp"
#include "pinocchio/parsers/sample-models.hpp"

#include "pinocchio/algorithm/compute-all-terms.hpp"
#include "pinocchio/tools/timer.hpp"

#include <iostream>

//-------------------------------- main function ----------------------------------------

int main(int argc, const char** argv)
{
    using namespace Eigen;
    using namespace se3;
    Eigen::Array3d m_x_COM;
    std::cout<<"hello world"<<std::endl;
    return 0;
}
```

When compiling, it gives me error like this:
```
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:167:14: error: ‘size_of_xpr_at_compile_time’ is not a member of ‘se3::internal’
In file included from /usr/include/eigen3/Eigen/Core:297:0,
                 from /opt/openrobots/include/pinocchio/spatial/fwd.hpp:22,
                 from /home/wul/workspace/Simulation/MjCBiped/main.cpp:7:
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:167:58: error: ‘::ret’ has not been declared
...
```

Is there an example of how to create a project using both Eigen and Pinocchio?

jcarpent, 2018-03-22:
You can check a typical `CMakeLists.txt` in https://github.com/stack-of-tasks/tsid. We use for that the cmake modules https://github.com/jrl-umi3218/jrl-cmakemodules which is pretty useful.

I think you must avoid using `-std=c++11` because Pinocchio on robotpkg has been compiled with this option.
The same applies for `-mavx`. You must first install Pinocchio using these options, or simply avoid to use it when prototyping your code.

You should the pkg-config (.pc file of pinocchio) to configure the cmake project.

longwoo, 2018-03-23:
@jcarpent  Thanks for the guide. It works.

jcarpent, 2018-03-23:
Good. I will add a minimal project with Pinocchio soon.

closed 2018-03-23 by jcarpent

jcarpent, 2018-03-26:
Hi @longwoo ,
I just created a typical starting repository of projects that use Pinocchio: https://github.com/stack-of-tasks/pinocchio-minimal

longwoo, 2018-03-27:
@jcarpent Nice. It is very useful which simplified the cmakelist I used before. But I find it is useless to add the ```fwd.hpp```  in ```main.cpp```, so I changed it like this.

```
#
# Copyright (c) 2018 CNRS
#
# This file is part of pinocchio-minimal
# pinocchio is free software: you can redistribute it
# and/or modify it under the terms of the GNU Lesser General Public
# License as published by the Free Software Foundation, either version
# 3 of the License, or (at your option) any later version.
# pinocchio is distributed in the hope that it will be
# useful, but WITHOUT ANY WARRANTY; without even the implied warranty
# of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU
# General Lesser Public License for more details. You should have
# received a copy of the GNU Lesser General Public License along with
# pinocchio If not, see
# <http://www.gnu.org/licenses/>.
CMAKE_MINIMUM_REQUIRED(VERSION 2.6)

INCLUDE(cmake/base.cmake)
INCLUDE(cmake/boost.cmake)

SET(PROJECT_NAME pinocchio-minimal)
SET(PROJECT_DESCRIPTION "Minimal example of a simple application depending on Pinocchio")
SET(PROJECT_URL "http://github.com/stack-of-tasks/pinocchio-minimal")

# Disable -Werror on Unix for now.
SET(CXX_DISABLE_WERROR True)
SET(CMAKE_VERBOSE_MAKEFILE True)

# ----------------------------------------------------
# --- DEPENDANCIES -----------------------------------
# ----------------------------------------------------
ADD_REQUIRED_DEPENDENCY("pinocchio >= 1.2.0")
ADD_REQUIRED_DEPENDENCY("tsid >= 1.0.0")
ADD_REQUIRED_DEPENDENCY("eigen3 >= 3.2.0")

# ----------------------------------------------------
# --- INCLUDE ----------------------------------------
# ----------------------------------------------------
# Add headers here if any
SET(${PROJECT_NAME}_HEADERS
    include/pinocchio-minimal/
  )
INCLUDE_DIRECTORIES ( ${${PROJECT_NAME}_HEADERS})

# ----------------------------------------------------
# --- EXECUTABLE -------------------------------------
# ----------------------------------------------------
SET(${PROJECT_NAME}_SOURCES
    src/main.cpp
  )
# Create the main executable
ADD_EXECUTABLE(main ${${PROJECT_NAME}_SOURCES})
# Link main to its dependencies 
PKG_CONFIG_USE_DEPENDENCY(main pinocchio tsid eigen3)
TARGET_LINK_LIBRARIES (main tsid)
# --- PACKAGING ----------------------------------------------------------------
PKG_CONFIG_APPEND_LIBS(${PROJECT_NAME})
```

So I can also use ```tsid``` lib which contain a useful tool ```tsid::robots::RobotWrapper```.