stack-of-tasks/pinocchio #450: colwise multiplication is much more faster than the direct Eigen multiplication?

longwoo, 2018-03-29:
I found a interesting thing.  The source code ```src/algorithm/center-of-mass.hxx``` line 260 says ``` I don't know why, but the colwise multiplication is much more faster than the direct Eigen multiplication```. So I had a test like this
```
        clock_t start = clock();
        for (long i=0; i<1000;++i){
            for (long k=0; k<robot->nv();++k)
                rb_data->Jcom.col(k) = rb_data->oMi[1].rotation() * rb_data->M.topRows<3> ().col(k);
        }

        clock_t ends = clock();
        std::cout <<"colwise multiplication Running Time : "<<(double)(ends - start)*1000/ CLOCKS_PER_SEC << std::endl;
        for (long i=0; i<1000;++i){
            rb_data->Jcom= rb_data->oMi[1].rotation() * rb_data->M.topRows<3> ();
        }

        clock_t ends1 = clock();
        std::cout <<"direct multiplication Running Time : "<<(double)(ends1 - ends)*1000/ CLOCKS_PER_SEC << std::endl;
```

It gives me 
```
colwise multiplication Running Time : 63.819
direct multiplication Running Time : 5.319
```

jcarpent, 2018-03-29:
Are you sure that your rb_data->M and rb_data->oMi[1] have been initilised with values through `crba` and `forwardKinematics` algo.

In addition, Eigen is doing here some intermediate allocation, because you are not using the noalias() strategy. Can you check that too?

longwoo, 2018-03-30:
@jcarpent here is my new test.
```
    tsid::robots::RobotWrapper * robot;
    se3::Data * rb_data;
    std::string urdfFileName = MODEL_DIR"/test_legs/test_legs.urdf";
    std::vector<std::string> package_dirs;
    package_dirs.push_back(MODEL_DIR"/test_legs");
    robot = new tsid::robots::RobotWrapper(urdfFileName,
                                           package_dirs,
                                           se3::JointModelFreeFlyer(),
                                           false);
    rb_data = new se3::Data(robot->model());

        Eigen::VectorXd pino_q = Eigen::Map<Eigen::ArrayXd>(&m_data->qpos[0], 19);//in mujoco floating base, qpos is (X,Y,Z,w,x,y,z)
        Eigen::VectorXd pino_v = Eigen::Map<Eigen::ArrayXd>(&m_data->qvel[0], 18);
        double t=0;//in pinocchio floating base, qpos is (X,Y,Z,x,y,z,w)
        t = pino_q(3);
        pino_q(3)=pino_q(4);
        pino_q(4)=pino_q(5);
        pino_q(5)=pino_q(6);
        pino_q(6)=t;
        robot->computeAllTerms(*rb_data , pino_q, pino_v);//tsid::robot::computeAllTerms
        std::cout <<"x_com in pinocchio === :\n "<<rb_data->com[0]<< std::endl;
        std::cout <<"x_com in mujoco === :\n "<<m_x_COM << std::endl;
        std::cout <<"x_vcom in pinocchio === :\n "<<rb_data->oMi[1].rotation().inverse()*rb_data->vcom[0]<< std::endl;
        std::cout <<"x_vcom in mujoco === :\n "<<m_v_COM << std::endl;
        std::cout <<"x_acom in pinocchio === :\n "<<rb_data->oMi[1].rotation().inverse()*rb_data->acom[0]<< std::endl;
        std::cout <<"x_acom in mujoco === :\n "<<m_vd_COM << std::endl;

        Eigen::MatrixXd J_com1;
        J_com1.resize(3,robot->nv());
        Eigen::MatrixXd J_com2;
        J_com2.resize(3,robot->nv());
        clock_t start = clock();
        for (long i=0; i<1000;++i){
            for (long k=0; k<robot->nv();++k)
                J_com1.col(k).noalias() =rb_data->oMi[1].rotation() * rb_data->M.topRows<3> ().col(k);
        }

        clock_t ends = clock();
        std::cout <<"colwise multiplication Running Time : "<<(double)(ends - start)*1000/ CLOCKS_PER_SEC << std::endl;
        for (long i=0; i<1000;++i){
            J_com2.noalias() = rb_data->oMi[1].rotation() * rb_data->M.topRows<3> ();
        }
        clock_t ends1 = clock();
        std::cout <<"direct multiplication Running Time : "<<(double)(ends1 - ends)*1000/ CLOCKS_PER_SEC << std::endl;
```

First, I use ```tsid::robot::computeAllTerms```,which contains
```
    void RobotWrapper::computeAllTerms(Data & data, const Vector & q, const Vector & v) const
    {
      se3::computeAllTerms(m_model, data, q, v);
      data.M.triangularView<Eigen::StrictlyLower>()
            = data.M.transpose().triangularView<Eigen::StrictlyLower>();
      // computeAllTerms does not compute the com acceleration, so we need to call centerOfMass
      se3::centerOfMass<true,true,true>(m_model, data, false);
      se3::framesForwardKinematics(m_model, data);
      se3::centerOfMass(m_model, data, q, v, Eigen::VectorXd::Zero(nv()));
    }
```
so I think ```rb_data->M``` and ```rb_data->oMi[1]``` have been initilised with values through ```crba``` and ```forwardKinematics``` algo.

Second, in pinocchio source code ```src/algorithm/center-of-mass.hxx```, I did't find noalias() strategy either.
```
  getJacobianComFromCrba(const Model & model, Data & data)
  {
    assert(model.check(data) && "data is not consistent with model.");
    const SE3 & oM1 = data.liMi[1];
    
    // As the 6 first rows of M*a are a wrench, we just need to multiply by the
    // relative rotation between the first joint and the world
    const SE3::Matrix3 oR1_over_m (oM1.rotation() / data.M(0,0));
    
    // I don't know why, but the colwise multiplication is much more faster
    // than the direct Eigen multiplication
    for (long k=0; k<model.nv;++k)
      data.Jcom.col(k) = oR1_over_m * data.M.topRows<3> ().col(k);
    return data.Jcom;
  }
```
And I add ```.noalias()``` in new test.
```
        for (long i=0; i<1000;++i){
            for (long k=0; k<robot->nv();++k)
                J_com1.col(k).noalias() =rb_data->oMi[1].rotation() * rb_data->M.topRows<3> ().col(k);
        }
```
It gives me 
```
colwise multiplication Running Time : 56.769
direct multiplication Running Time : 4.397
```
Both of them reduce computation time.Do I misunderstand the noalias() strategy ? Or I have to add something in Cmakelist?

jcarpent, 2018-03-30:
It is possible that your compiler is too smart and do not do the complete loop (for i = 0 until 1000), because it realizes that the intermediate computations are not useful.
Writing precise benchmark is pretty hard, and I really advise to see the ones of Eigen which are very atomic.
Just to be check what you mentioned, you can run twice the `timings` benchmark inside the benchmark folder of Pinocchio. One one side, you just complete the standard bench. On the other side, you modify the line 260 of `src/algorithm/center-of-mass.hxx` and set your code. You will be then able to check which operation is better?

longwoo, 2018-03-30:
@jcarpent It is amazing! I test it in source code 10000 times like this 
```
  getJacobianComFromCrba(const Model & model, Data & data)
  {
    assert(model.check(data) && "data is not consistent with model.");
    const SE3 & oM1 = data.liMi[1];

    // As the 6 first rows of M*a are a wrench, we just need to multiply by the
    // relative rotation between the first joint and the world
    const SE3::Matrix3 oR1_over_m (oM1.rotation() / data.M(0,0));
    
    // I don't know why, but the colwise multiplication is much more faster
    // than the direct Eigen multiplication
    clock_t start = clock();
    for(int i=0;i<10000;i++){
    	for (long k=0; k<model.nv;++k){
    		data.Jcom.col(k) = oR1_over_m * data.M.topRows<3> ().col(k);
    	}
    }
    clock_t ends = clock();
    std::cout <<"colwise multiplication Running Time : "<<(double)(ends - start)*1000/ CLOCKS_PER_SEC << std::endl;
    for(int i=0;i<10000;i++){
    	data.Jcom = oR1_over_m * data.M.topRows<3> ();
    }
    clock_t ends1 = clock();
    std::cout <<"direct multiplication Running Time : "<<(double)(ends1 - ends)*1000/ CLOCKS_PER_SEC << std::endl;
    return data.Jcom;
```
And it gives me 
```
colwise multiplication Running Time : 2.559
direct multiplication Running Time : 3.573
```
Eigen is a good thing!

jcarpent, 2018-03-30:
@longwoo . Cool! So this test confirms the intuition of the comment inside the code.
I will close the issue now. Do not hesitate to reopen it later if needed.

closed 2018-03-30 by jcarpent

jmirabel, 2018-03-31:
Since you do benchmarks of Eigen, could you try to set the two last template arguments of `Eigen::Matrix< _Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols >` (`MaxColsAtCompileTime` and `MaxRowsAtCompileTime`) to some values in this two places :
https://github.com/stack-of-tasks/pinocchio/blob/5842ba5ded84b5c39f2e06bd9b1348c6c9f7e254/src/multibody/model.hpp#L398
and
https://github.com/stack-of-tasks/pinocchio/blob/5842ba5ded84b5c39f2e06bd9b1348c6c9f7e254/src/multibody/model.hpp#L476

and then run the same benchmark ?

longwoo, 2018-04-02:
@jmirabel 

1.  If I only change line 398 into 
```
Eigen::Matrix<double,Eigen::Dynamic,Eigen::Dynamic, 0, 50,50> M;//todo
```
It gives me 
```
colwise multiplication Running Time : 2.24
direct multiplication Running Time : 0.803
```
2.  If I only change line 476 into 
```
Eigen::Matrix<double,3 ,Eigen::Dynamic, 0, 3,50> Jcom;//todo
```
It gives me 
```
colwise multiplication Running Time : 2.612
direct multiplication Running Time : 2.852
```
3. If I change both of them .It gives me 
```
colwise multiplication Running Time : 4.177
direct multiplication Running Time : 1
```

jmirabel, 2018-04-06:
Thanks for reporting the results. I do not really understand the reason for such great differences though...