stack-of-tasks/pinocchio #451: Installation procedure through robotpkg

andreadelprete, 2018-04-12:
I think I found a mistake in the installation procedure described on [Pinocchio's website](https://stack-of-tasks.github.io/pinocchio/download.html). In the section **Add robotpkg PPA**, point 2, it tells to add _http://robotpkg.openrobots.org/packages/debian/pub/dists_. However, in the [robotpkg website](http://robotpkg.openrobots.org/binary.html), it tells to add _http://robotpkg.openrobots.org/packages/debian/pub_. On my machine the first method does not work, while the second does, so I guess there is a mistake on Pinocchio's website.

jcarpent, 2018-04-12:
Hi Andrea,
Indeed, the procedure on the website of Pinocchio is wrong and should follow the instructions given by https://github.com/stack-of-tasks/pinocchio/wiki/Installation.
Thanks for reporting it here. 
@nim65s Can you make the website of Pinocchio to be synchronized with the wiki of github?
Thanks in advance.

nim65s, 2018-04-12:
I clearly forgot to update this page when I updated the template of the website, sorry.

Anyway, why do we even have this information duplicated on the website and the wiki, and even on https://github.com/stack-of-tasks/pinocchio-tutorials/tree/master/tp0_install ?

Please, let's choose only one of these 3, update it, remove the other and replaces them by links to the main one. I suggest keeping the website.

nim65s, 2018-04-12:
Got it, the website was using javascript to fetch informations from the wiki and display it, and I removed this.

Avoiding unnecessary javascript (especially if they come from an external untrusted source) on our websites would be really better, so I still suggest to display this information in only one place

jcarpent, 2018-04-12:
I do agree. The idea of putting on the wiki of Github is for simplicity of updating the content.
But, if you prefer, we can put everything on gh-pages.

commit 4a846099 (2018-04-20): Merge pull request #455 from nim65s/gh-pages

update installation procedure from the wiki, fix #451
commit 945a92f8 (2018-04-20): update installation procedure from the wiki, fix #451
closed 2018-04-20 by jcarpent