stack-of-tasks/pinocchio #453: computeAllTerms crash on model with only one joint

nim65s, 2018-04-16:
Here is a test:

```py
import pinocchio as se3
from pinocchio.utils import *

nb_joints = 1

model = se3.Model.BuildEmptyModel()
for i in range(nb_joints):
    model.addJoint(i, se3.JointModelRY(), se3.SE3.Identity(), 'joint_%i' % i)

data = model.createData()

se3.computeAllTerms(model, data, zero(model.nq), zero(model.nv))
```

Setting `nb_joints` to 2 or more works, but for 1 I get:

```
Traceback (most recent call last):
  File "pinocchio_test.py", line 14, in <module>
    se3.computeAllTerms(model, data, zero(model.nq), zero(model.nv))
Boost.Python.ArgumentError: Python argument types in
    pinocchio.libpinocchio_pywrap.computeAllTerms(Model, Data, matrix, matrix)
did not match C++ signature:
    computeAllTerms(se3::Model Model, se3::Data {lvalue} Data, Eigen::Matrix<double, -1, 1, 0, -1, 1> Configuration q (size Model::nq), Eigen::Matrix<double, -1, 1, 0, -1, 1> Velocity v (size Model::nv))
```

jcarpent, 2018-04-16:
This problem is coming from eigenpy in fact. I will try to fix it.

jcarpent, 2018-04-16:
This if fixed by stack-of-tasks/eigenpy#28.

closed 2018-04-16 by jcarpent