stack-of-tasks/pinocchio #464: Installing .pyc files

nim65s, 2018-05-25:
Hi,

In https://github.com/stack-of-tasks/pinocchio/blob/master/bindings/python/CMakeLists.txt#L214 we try to install .pyc files.

First of all, I don't understand why.
Does someone have an explanation ?

Then, in python 3.2+, [PEP 3147](https://www.python.org/dev/peps/pep-3147/) states that compiled files are no more `$filename.pyc`, but `__pycache__/$filename.<magic>.pyc` with `<magic>` something like `cpython-35`, so the project is not building, because cmake is still looking for `$filename.pyc`.

jmirabel, 2018-05-25:
I guess it is mostly a copy-paste from https://github.com/jrl-umi3218/jrl-cmakemodules/blob/d22de8c53c3507df785b1fb3ab260d9fdfb65233/python.cmake#L183

jmirabel, 2018-05-25:
As of now, I could not find any pros and cons. You have found an important cons.

jcarpent, 2018-05-25:
It is just an historical reason.
You can remove this pass inside the CMakeLists.txt.

nim65s, 2018-05-25:
will do :)

jmirabel, 2018-05-25:
Can you open an issue in https://github.com/jrl-umi3218/jrl-cmakemodules as well ?

jcarpent, 2018-05-26:
Solved by #465

commit 8504b380 (2018-05-26): [CMake] don't compile .py files nor try to install them, fix #464
commit 4a555b7d (2018-06-08): [CMake] don't compile .py files nor try to install them, fix #464
closed 2018-05-26 by jcarpent