stack-of-tasks/pinocchio #467: Visualization with Gepetto

mpetriaux, 2018-05-30:
I'm trying to visualize a robot using gepetto viewer and pinocchio.
I'm using pinocchio version 1.2.8  and gepetto viewer version 1.1.0

The following code sample does not work:
```python
from pinocchio.robot_wrapper import RobotWrapper
import pinocchio as pnc

URDF_PATH = "my_urdf_path"

robot = RobotWrapper(URDF_PATH, "my_meshes_package_path", pnc.JointModelFreeFlyer())
robot.initDisplay("world/my_robot", loadModel=True)
```

I previously ran `gepetto-viewer-server` in another terminal.

Still I get the following error: 
```
Error while starting the viewer client. 
Check wheter gepetto-viewer is properly started
```

Is this the correct way to display the robot? What can I do to correct it?

jcarpent, 2018-05-30:
Do you start corba omninames?
Can you try with gepetto-gui?

aelkhour, 2018-05-30:
@jcarpent I'm not familiar with gepetto-gui, what is the difference with respect to gepetto-viewer-server?

nmansard, 2018-05-30:
It is an equivalent software based on QT which has proven to be slightly more reliable that GVS in the past.

aelkhour, 2018-05-30:
We have an osg install without QT plugins, so it's going to be tedious to test with gepetto-gui.
Is geppeto-viewer-server deprecated?

omniNames is already started when running the test described by @mpetriaux. I encoutered the same issue.

jmirabel, 2018-05-30:
- Does `gepetto-viewer-server` print an error message ?
- In robot_wrapper.py, in method `RobotWrapper.initDisplay`, could you catch errors of type `gepetto.Error` and print them ? @jcarpent I think it should be changed as the potentially helpful error message is hidden.

> We have an osg install without QT plugins, so it's going to be tedious to test with gepetto-gui.

Did you manually deactivate it ? By default, it is activated. If you reconfigure OSG, use cmake flag `-DDESIRED_QT_VERSION=4`

jmirabel, 2018-05-30:
Then, for `gepetto-gui`, you have no other options than recompiling.

For `gepetto-viewer-server`, did you get the opportunity to answer my two other questions ?

aelkhour, 2018-05-30:
Thanks @jmirabel, we'll get back to you regarding our tests asap.

mpetriaux, 2018-06-01:
Thank you all for your answers.

I've been digging in initDisplay method.

First, this line `scene_l = gui.getSceneList()` doesn't works on our gepetto version, getSceneList() doesn't exist.

This doesn't seem so bad, if I just skip this line and assume the scene doesn't exist I can continue until `self.loadDisplayModel()`

Same thing here, method `gui.nodeExists()` is not defined but I can skip the check (l. 245) and continue.

Then if I continue step by step I end up with a robot displayed.

Meshes are ill-placed in the viewer (this might be an internal issue, I'll keep you informed) but at least they are here.

jcarpent, 2018-06-01:
@mpetriaux Just to be sure, which version of gepetto are you currently using?
Why to not use the last version on the robotpkg repository?

aelkhour, 2018-06-01:
As mentioned in the issue by @mpetriaux, we use gepetto-viewer v1.1.0, so we are indeed a bit late with respect to the latest releases.

We can try to update to the latest stable release of gepetto-viewer.

mpetriaux, 2018-06-01:
So I'm trying to update to latest releases of gepetto-viewer, gepetto-viewer-corba and eigenpy.

But then I need osg-Qt.

I tried to pass the CMake flag you advised @jmirabel but it's not used by the project.

What version of OpenSceneGraph are you using?

jcarpent, 2018-06-01:
@mpetriaux 
Do you know that you can install all these sotfwares through robotpkg ppa (binairies).
You can use the same instructions than for Pinocchio:
https://github.com/stack-of-tasks/pinocchio/wiki/Installation

You just need to add the ppa of robotpkg as explained in the installation instructions of Pinocchio and run:
`sudo apt-get install robotpkg-gepetto-viewer-corba`
It will install gepetto-viewer-corba and gepetto-gui with all their dependencies.

jmirabel, 2018-06-01:
If @jcarpent solution does not suit your needs (which is the easiest):

For OpenSceneGraph, you have two solutions (working at least on Ubuntu 14.04 and 16.04):
- `apt-get install libopenscenegraph-dev`,
- installation version 3.2 or 3.4 from source. If you choose this option, make sure you have Qt installed and is found by OSG, then add the CMake flag I mentioned earlier.

aelkhour, 2018-06-02:
I tried out @jcarpent's way and managed to launch gepetto-gui along with @mpetriaux's script. The URDF geometries (both visual and collision) are loaded in the scene, but for some reason they appear to be "black". The mesh files are stored in STL format, and used to be displayed fine in previous versions of gepetto-viewer. See below picture for comparison between gepetto-gui and meshlab for the same mesh file.

[image]

For the record, the corresponding urdf part is:

```xml
    <visual>
      <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0"/>
      <geometry>
        <mesh filename="package://my_meshes/MyLink.STL"/>
      </geometry>
      <material name="">
        <color rgba="0.752941176471 0.752941176471 0.752941176471 1.0"/>
      </material>
    </visual>
```

jmirabel, 2018-06-03:
This is due to a bug in the STL plugin of OSG in version 3.2. The bug is fixed in 3.4.

From binary packages, the only option (though I never tried) is to convert them in another format (using meshlab for instance).

If you do not want to modify the URDF file, you can do:
- convert `file.stl` to `file.obj` (OBJ, DAE or anything you like)
- run command `osgconv file.obj file.stl.osgb`.

Explanation: when gepetto-gui opens mesh `mesh.ext`, it first check whether there exists some `mesh.ext.(osg|osgb|osg2)`. If such a file exists, it opens it instead of `mesh.ext`. This is useful when loading big meshes because OSG native formats (especially `osgb`, b for binary) are a lot faster to open than text file formats. In the above scenario, `file.stl.osgb` will be used instead of `file.stl`.

mpetriaux, 2018-06-04:
So I've found a patch to make pinocchio 1.2.8 work with our gepetto viewer version (1.1.0).

I removed calls to `getSceneList` and `nodeExists`.

Then I changed the quaternion convention in `XYZQUATToViewerConfiguration` and call this method. 

For me the issue is solved. 

I think we will soon update to latest versions of eigenpy and gepetto-viewer which will be a more sustainable solution.

Thank you for your help!

@aelkhour I let the issue open in case you have additional questions one the STL files topic, feel free to close.

aelkhour, 2018-11-02:
@jcarpent Assuming we use the robotpkg PPA to install everything, does the above script work with pinocchio v2.0.0-alpha and gepetto-gui?

jcarpent, 2018-11-02:
They should in practice.

aelkhour, 2018-11-04:
I installed pinocchio v2.0.0-alpha on Ubuntu 18.04, and installed gepetto-viewer-corba from the robotpkg PPA.

I launched gepetto-gui and ran the following simple script

```python
import pinocchio

urdf_path = 'pendulum.urdf'

robot = pinocchio.RobotWrapper()
robot.initFromURDF(urdf_path, '.', pinocchio.JointModelFreeFlyer())
robot.initDisplay('python-pinocchio', 'world', loadModel=True)
```

I get an empty scene in gepetto-gui, with the following log:

```
List of GroupNodes :
   hpp-gui
 does not exist.: iostream error
 does not exist.: iostream error
 does not exist.: iostream error
 does not exist.: iostream error
Node "world/pinocchio/visuals/world_0" does not exist.
Node "world/pinocchio/visuals/PendulumArm_0" does not exist.
Node "world/pinocchio/visuals/PendulumIMU_0" does not exist.
Node "world/pinocchio/visuals/PendulumMass_0" does not exist.
```

Am I doing something wrong? Maybe I'm not using RobotWrapper the way it's supposed to be used?

`pendulum.urdf` is as follows:

```xml
<?xml version="1.0" ?>
<robot name="pendulum">
    <link name="world">
		<visual>
			<origin xyz="0 0 0" rpy="0 0 0" />
			<geometry>
				<sphere radius="0.005"/>
			</geometry>
			<material name="">
				<color rgba="1.0 1.0 1.0 1.0"/>
			</material>
		</visual>
		<inertial>
			<origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0"/>
			<mass value="1000.0"/>
			<inertia ixx="0.0" ixy="0.0" ixz="0.0" iyy="0.0" iyz="0.0" izz="0.0"/>
		</inertial>
    </link>

   <joint name="PendulumJoint" type="revolute">
      <origin xyz="0.0 0.0 0.0" rpy="0 0 0"/>
      <axis xyz="1 0 0"/>
      <parent link="world"/>
      <child link="PendulumArm"/>
      <limit effort="0" velocity="0"/>
    </joint>

    <link name="PendulumArm">
		<visual>
			<origin xyz="0 0 0.25" rpy="0 0 0" />
			<geometry>
				<box size="0.01 0.01 1.0"/>
			</geometry>
			<material name="">
				<color rgba="1.0 1.0 1.0 1.0"/>
			</material>
		</visual>
		<inertial>
			<origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0"/>
			<mass value="0.0001"/>
			<inertia ixx="0.00001" ixy="0.0" ixz="0.0" iyy="0.00001" iyz="0.0" izz="0.00001"/>
		</inertial>
    </link>

    <joint name="PendulumMassJoint" type="fixed">
      <origin xyz="0.0 0.0 1.0" rpy="0 0 0"/>
      <parent link="PendulumArm"/>
      <child link="PendulumMass"/>
    </joint>
    <link name="PendulumMass">
		<visual>
			<origin xyz="0 0 0" rpy="0 0 0" />
			<geometry>
				<sphere radius="0.06"/>
			</geometry>
			<material name="">
				<color rgba="1.0 1.0 1.0 1.0"/>
			</material>
		</visual>
		<inertial>
			<origin xyz="0.1 0.0 0.0" rpy="0.0 0.0 0.0"/>
			<mass value="1.0"/>
			<inertia ixx="0.001" ixy="0.0" ixz="0.0" iyy="0.001" iyz="0.0" izz="0.001"/>
		</inertial>
    </link>
    <joint name="IMUJoint" type="fixed">
      <origin xyz="0.0 0.0 0.75" rpy="0 0 0"/>
      <parent link="PendulumArm"/>
      <child link="PendulumIMU"/>
    </joint>
    <link name="PendulumIMU">
		<visual>
			<origin xyz="0 0 0" rpy="0 0 0" />
			<geometry>
				<sphere radius="0.025"/>
			</geometry>
			<material name="">
				<color rgba="1.0 1.0 1.0 1.0"/>
			</material>
		</visual>
		<inertial>
			<origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0"/>
			<mass value="0.0001"/>
			<inertia ixx="0.0" ixy="0.0" ixz="0.0" iyy="0.0" iyz="0.0" izz="0.0"/>
		</inertial>
    </link>
</robot>
```

nmansard, 2018-11-05:
The pinocchio implementation of Gepetto-viewer does not allow yet to display urdf files including simple geometries. 
To do so, we would need to wrap FCL objects, in order to recover the geomtries parameters, and then to add the dedicated code to call Gepetto-Viewer routines in the RobotWrapper class, display method. See #548

aelkhour, 2018-11-05:
> The pinocchio implementation of Gepetto-viewer does not allow yet to display urdf files including simple geometries.

Strange, this worked for at least pinocchio v1.2.8. Also, I'm pretty sure I have the same problem when loading meshes, which is our real issue (I will double check).