stack-of-tasks/pinocchio #470: Extracting subtree for independent computation

olivier-stasse, 2018-06-06:
Dear all,
I need to test and port old fashion control algorithm. They need to perform computation on subtree of a humanoid robot. Typically it would consist in making inverse kinematic computations of the left and right legs seperately.
I found subtrees in the structure of pinocchio but not a proper way to extract path of joints as it is done in other libraries.
Thanks for your help.

cmastalli, 2018-06-06:
I'm working in similar topic, i.e. computation of kinematic quantities (FK and IK) at the subtree level and other functions. Those functions are implemented in a library on top of Pinocchio, with the possibility of having Python bindings too. However, I'm currently testing the code (i.e. creating unittest).

The reason to implement them in an another library is due to the fact that these functions need another level of semantic description needed in motion control or planning. Just to give you a quick taste of this, imagine that you want to compute the FK of only feet branches.

This might interested topic of discussion for other people. @olivier-stasse doesn't hesitate to contact me if you're interested.

jcarpent, 2019-11-19:
@olivier-stasse There is now the notion of subtrees in Pinocchio, allowing to extract all the joints ids of this subtree. I plan to add the `forwardKinematics` of subtree in the very next future. Are you still interested by this feature?

Neotriple, 2020-02-18:
Just wanted to chime in to say that I would definitely love this feature. Pretty similar to what @cmastalli was referring to, having a subtree of the feet would be a really nice feature.

rshum19, 2021-02-03:
Hi @jcarpent ,  

I have a mobile manipulator model (robot base + robot arm). I'm trying to extract the Lagrangian dynamics for only the robot arm. Is there a way of extracting  just the robot arm as a subtree and using function like crba() and computeCoriolisMatrix()? 

thanks,
Roberto

jcarpent, 2021-02-03:
If you want to isolate the robot arm as a model, you have the function: `buildReducedModel` which does the work and then allows to reuse all the algorithms as usual.
Currently, the features are not yet available, as we are mostly lacking manpower.

rshum19, 2021-02-03:
`buildReducedModel` seem to be what I'm looking for. However, I didn't quite understand :

> Currently, the features are not yet available, as we are mostly lacking manpower.

Does this mean that the `buildReducedModel` function has been defined but not yet implemented or that the feature described to perform `forwardKinematics` in subtrees is not yet available?

jcarpent, 2021-02-03:
`buildReducedModel` works perfectly well. What I meant is that working with subtrees is not yet possible

rshum19, 2021-02-03:
I see great thank you for the clarification!