stack-of-tasks/pinocchio #476: [Support] Q&A

nim65s, 2018-06-18:
Hi,

This tracker is useful to maintain a todo-list and report bugs, but it seems that we need a Q&A to let our users ask general questions. For this, a general website like [Stack 
Overflow](https://stackoverflow.com/) seems nice.

On this website, we can find a few tags like robot & robotics, but they suggest to use another website from Stack Exchange: http://robotics.stackexchange.com/

There, we can find more precise tags, like humanoid, legged, dynamics, kinematics, inverse-kinematics, forward-kinematics, etc.

If you want to take a look, you should start by the [tour](https://robotics.stackexchange.com/tour).

I suggest that we use this site, and subscribe to the humanoid and/or legged tags, and let our users ask their questions there.

But we must keep in mind that on this website, questions have to be relevant and well formulated, and that they are looking for comprehensive answers that provide some explanation and context.

nim65s, 2018-06-18:
An alternative to Stack Exchange would be to use [ROS Answers](https://answers.ros.org/questions/), but we should put Pinocchio in ROS first :)

jcarpent, 2018-07-03:
Thanks for this list of solutions.
I recently came to this alternative solution: https://gitter.im

@nim65s What do you think?

nim65s, 2018-07-03:
gitter is a Instant Messaging solution, and has nothing to do with Q&A.

It is a closed-source direct concurrent to slack (closed-source), rocket.chat, mattermost, & matrix (open-source) and a billion others ([mandatory xkcd](https://xkcd.com/1810/)).

This could be discussed in a separate issue, but we can also provide support over an IM solution.
I think this is something we must do, and I already explained that to the team multiple times. For now, we have a mattermost (framateam) group, and already provide support there to those who use this: https://framateam.org/gepetto/channels/pinocchio

Now, Mattermost is not perfect, so we can discuss that.

jcarpent, 2018-07-03:
Ok. I see your point.

By the way, is it possible to integrate Mattermost inside Github. I mean, like a plug-in.
I'm trying to figure out this feature, but I cannot fin any hint on Internet.

nim65s, 2018-07-03:
It is possible, but it works really better with gitlab. In fact, a mattermost instance is automatically provided with a gitlab instance, and they work together. I think we should discuss this in the next HPP-Meeting, if you are here.

olivier-stasse, 2020-02-10:
Why not running an Askbot server as it is done with ROS ?

kozakromch, 2021-08-10:
Hello, Guys. I hope that I am asking a question in the right section, if not, please tell me where it is worth asking questions of this kind. I'm sorry to bother you. Thank you for such a wonderful library.

I do not know how to solve the following problem:
I have an anthropomorphic model. all joints are spherical or revolute. Pelvis -- free floating, No loops. Just a tree topology. I know the global coordinates of each bone.  Also I know that this global coordinate get from correct model configuration, so definitely one model configuration q, that maps to my global coordinates, exists. How can I restore the model configuration q based on these global coordinates?

I had a simple plan - to start from pelvis and move down the hierarchy and calculate the local coordinates, and then fill in the model configuration q . But I can't find anywhere how to map joint local coordinate and index in q. And which coordinate are in model configuration q and which not. 

This task resembles a simplified version of inverse kinematics, but, unfortunately, I could not find any implementation.
I have found only two examples in which the inverse kinematics problem is solved through a forward kinematics, but it seems to me that this is too difficult for my case.

Thanks in advance

jcarpent, 2021-08-10:
This is not the right place indeed. There is a place called discussions in GitHub https://github.com/stack-of-tasks/pinocchio/discussions
To quickly answer your question, you have to solve an IK problem to retrieve the best configuration estimate.
We do not provide an algo for that, but from the tutorials and the examples, this is pretty straightforward.

closed 2021-08-24 by jcarpent