stack-of-tasks/pinocchio #511: Const correctness in boost::fusion::vector for ArgsType & Boost versions

nim65s, 2018-09-10:
Hi,

If we consider this simple exemple:
```
#include <boost/fusion/container/vector.hpp>

struct Base
{
    typedef boost::fusion::vector<const unsigned long, int> ArgsType;
    ArgsType args;

    Base(ArgsType args): args(args) {};
};

int main()
{
    Base::ArgsType args(3, 4);
    Base instance(args);
    return 0;
}
```

It compiles on Boost 1.64, 1.65, doesn't n 1.66 & 1.67, and compiles again in 1.68, as you can see here: https://godbolt.org/z/aHrEBe

When it doesn't, the issue is with the `const` keyword:
```
/opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/container/vector/detail/deref_impl.hpp:47:49: error: binding reference of type 'boost::fusion::extension::deref_impl<boost::fusion::vector_iterator_tag>::apply<boost::fusion::vector_iterator<boost::fusion::vector<const long unsigned int, int>, 0> >::type' {aka 'long unsigned int&'} to 'const long unsigned int' discards qualifiers
```

I will dig a bit, but I think this is an issue in Boost, so I suggest to simply avoid those versions.

jcarpent, 2018-09-10:
And with const ref?

jcarpent, 2018-09-10:
By the way, can you shown the error provided by gcc?

nim65s, 2018-09-10:
With `typedef boost::fusion::vector<const unsigned long, int> ArgsType;`, it works in all my tests, from 1.64 to 1.68.

Without it, here is the gcc 8.2 error:
```
In file included from /opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/container/vector/vector_iterator.hpp:12,

                 from /opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/container/vector/detail/begin_impl.hpp:11,

                 from /opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/container/vector/vector.hpp:31,

                 from /opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/container/vector.hpp:12,

                 from <source>:1:

/opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/container/vector/detail/deref_impl.hpp: In instantiation of 'static constexpr boost::fusion::extension::deref_impl<boost::fusion::vector_iterator_tag>::apply<Iterator>::type boost::fusion::extension::deref_impl<boost::fusion::vector_iterator_tag>::apply<Iterator>::call(const Iterator&) [with Iterator = boost::fusion::vector_iterator<boost::fusion::vector<const long unsigned int, int>, 0>; boost::fusion::extension::deref_impl<boost::fusion::vector_iterator_tag>::apply<Iterator>::type = long unsigned int&]':

/opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/iterator/deref.hpp:63:32:   required from 'constexpr typename boost::fusion::result_of::deref<Iterator>::type boost::fusion::deref(const Iterator&) [with Iterator = boost::fusion::vector_iterator<boost::fusion::vector<const long unsigned int, int>, 0>; typename boost::fusion::result_of::deref<Iterator>::type = long unsigned int&]'

/opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/iterator/deref.hpp:71:29:   required from 'constexpr typename boost::fusion::result_of::deref<Iterator>::type boost::fusion::operator*(const boost::fusion::iterator_base<Iterator>&) [with Iterator = boost::fusion::vector_iterator<boost::fusion::vector<const long unsigned int, int>, 0>; typename boost::fusion::result_of::deref<Iterator>::type = long unsigned int&]'

/opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/container/vector/vector.hpp:122:41:   required from 'constexpr typename boost::fusion::vector_detail::result_of::forward_at_c<Sequence, N>::type boost::fusion::vector_detail::forward_at_c(Sequence&&) [with int N = 0; Sequence = boost::fusion::vector<const long unsigned int, int>&; typename boost::fusion::vector_detail::result_of::forward_at_c<Sequence, N>::type = long unsigned int&]'

/opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/container/vector/vector.hpp:202:46:   required from 'constexpr boost::fusion::vector_detail::vector_data<std::integer_sequence<long unsigned int, _Idx ...>, T ...>::vector_data(boost::fusion::vector_detail::each_elem, Sequence&&) [with Sequence = boost::fusion::vector<const long unsigned int, int>&; Sequence_ = boost::fusion::vector<const long unsigned int, int>; <template-parameter-2-3> = void; long unsigned int ...I = {0, 1}; T = {const long unsigned int, int}]'

/opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/container/vector/vector.hpp:306:75:   required from 'constexpr boost::fusion::vector<T>::vector(Sequence&&) [with Sequence = boost::fusion::vector<const long unsigned int, int>&; <template-parameter-2-2> = void; T = {const long unsigned int, int}]'

<source>:8:35:   required from here

/opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/container/vector/detail/deref_impl.hpp:47:49: error: binding reference of type 'boost::fusion::extension::deref_impl<boost::fusion::vector_iterator_tag>::apply<boost::fusion::vector_iterator<boost::fusion::vector<const long unsigned int, int>, 0> >::type' {aka 'long unsigned int&'} to 'const long unsigned int' discards qualifiers

                     return i.vec.at_impl(index());

                                                 ^

Compiler returned: 1
```

And with clang 6.0:
```
In file included from <source>:1:

In file included from /opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/container/vector.hpp:12:

In file included from /opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/container/vector/vector.hpp:31:

In file included from /opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/container/vector/detail/begin_impl.hpp:11:

In file included from /opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/container/vector/vector_iterator.hpp:12:

/opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/container/vector/detail/deref_impl.hpp:47:28: error: binding value of type 'const unsigned long' to reference to type 'unsigned long' drops 'const' qualifier

                    return i.vec.at_impl(index());

                           ^~~~~~~~~~~~~~~~~~~~~~

/opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/iterator/deref.hpp:63:28: note: in instantiation of member function 'boost::fusion::extension::deref_impl<boost::fusion::vector_iterator_tag>::apply<boost::fusion::vector_iterator<boost::fusion::vector<const unsigned long, int>, 0> >::call' requested here

        return deref_meta::call(i);

                           ^

/opt/compiler-explorer/libs/boost_1_67_0/boost/fusion/iterator/deref.hpp:71:24: note: in instantiation of function template specialization 'boost::fusion::deref<boost::fusion::vector_iterator<boost::fusion::vector<const unsigned long, int>, 0> >' requested here

        return fusion::deref(i.cast());

                       ^

[... cut ...] unsigned long, int>, void>' requested here

            : base(vector_detail::each_elem(), std::forward<Sequence>(seq))

              ^

<source>:8:26: note: in instantiation of function template specialization 'boost::fusion::vector<const unsigned long, int>::vector<boost::fusion::vector<const unsigned long, int> &, void>' requested here

    Base(ArgsType args): args(args) {};

                         ^

1 error generated.

Compiler returned: 1
```

So should we use const ref everywhere ? :)

jcarpent, 2018-09-10:
Certainly, but it is weird, because this error is not appearing on OS X with boost form 1.63 up to 1.67.
I have other modifications that are not compatible with such fix. I would prefer to wait the next release of Pinocchio 2.0.0 with full templatization to fix it.

nim65s, 2018-09-10:
the patch is here:
https://github.com/nim65s/robotpkg/blob/master/math/pinocchio/patches/patch-ab

jviereck, 2018-10-09:
Hi folks, I came across this issue. For me it showed up when I started to use the pinocchio headers for another (macOS) project, that used C++11 features. I provided my fixes here:

  https://github.com/jviereck/pinocchio/commit/250aada7d808bc2c4a612e23793bc735a4760c96

jcarpent, 2018-10-10:
This problem is coming from boost itself.
Boost has two different C++ signatures if you are compliant with C++98 or C++11 .
The welcome solution is to not provide a library with Pinocchio. This will be a part of the very next release.

traversaro, 2018-10-10:
@jcarpent You are probably aware of this, but a relevant discussion on the boost side is in https://groups.google.com/forum/#!topic/boost-developers-archive/EWG5NVOZo_g (found via https://github.com/Microsoft/vcpkg/issues/4004#issuecomment-414188408).

jcarpent, 2018-10-10:
@traversaro Thank you a lot for pointing out these interesting discussions.

jcarpent, 2018-10-29:
@nim65s Can you check that branch devel is solving this issue?

nim65s, 2018-10-30:
It does not ; I will provide a fix soon.

nim65s, 2018-11-22:
So… It's still not working in #601, and I can't understand why.

This issue occurs in Boost 1.66 & 1.67, with any other c++ standard than c++98.

Then, if nobody else provide a fix, I guess that we should drop support for this c++11 and more on Boost 1.66 & 1.67. 

In the list of distributions supported by robotpkg, this will affect fedora 28 & 29 (boost 1.66) and netbsd8 (which provides 1.65, 1.66, 1.67 and 1.68)

jcarpent, 2018-11-22:
C++11 is mandatory for lib generation and code source generation. So we cannot drop C++11 support.

jcarpent, 2018-11-22:
By the way, a simple fix would be to use pointers instead of references.

nmansard, 2018-11-22:
@nim65s is suggesting to drop support for [C++11] AND [boost 1.66-7] (not OR). I believe that this is acceptable. Possibly, you may document this problem by adding, using one way or the other, the tests you did to try to solve this issue.

nim65s, 2018-11-22:
Yes, this is what I meant. 
Affected users (on fedora) will still be able to install a more recent version of Boost.

jcarpent, 2018-11-22:
I think it would be worth to test the pointer version. @nim65s It requires just few minutes for you to test.

jcarpent, 2019-01-09:
Fixed by #635.

nim65s, 2019-01-09:
Sorry about that, but it's still not building on Fedora 28:
```
[ 72%] Building CXX object unittest/CMakeFiles/joint-generic.dir/joint-generic.cpp.o
cd /home/gsaurel/fedora/robotpkg/math/pinocchio/work.fluffy64-fedora28/pinocchio-1.3.3/unittest && /usr/bin/g++  -DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_WITH_CPPAD_SUPPORT -DPINOCCHIO_WITH_HPP_FCL -DPINOCCHIO_WITH_URDFDOM -I/home/gsaurel/fedora/robotpkg/math/pinocchio/work.fluffy64-fedora28/pinocchio-1.3.3 -I/home/gsaurel/fedora/robotpkg/math/pinocchio/work.fluffy64-fedora28/pinocchio-1.3.3/include -isystem /usr/include/eigen3 -isystem /home/gsaurel/fedora/install/include -isystem /usr/include/assimp -I/home/gsaurel/fedora/robotpkg/math/pinocchio/work.fluffy64-fedora28/pinocchio-1.3.3/unittest  -pedantic -Wno-long-long -Wall -Wextra -Wcast-align -Wcast-qual -Wformat -Wwrite-strings -Wconversion    -O3 -DNDEBUG    -DBOOST_TEST_DYN_LINK -DBOOST_TEST_MODULE=joint_genericTest -DFCL_USE_ASSIMP_UNIFIED_HEADER_NAMES -DFCL_HAVE_OCTOMAP= -o CMakeFiles/joint-generic.dir/joint-generic.cpp.o -c /home/gsaurel/fedora/robotpkg/math/pinocchio/work.fluffy64-fedora28/pinocchio-1.3.3/unittest/joint-generic.cpp
In file included from /usr/include/boost/fusion/container/vector/vector_iterator.hpp:12,
                 from /usr/include/boost/fusion/container/vector/detail/begin_impl.hpp:11,
                 from /usr/include/boost/fusion/container/vector/vector.hpp:31,
                 from /usr/include/boost/fusion/container/generation/make_vector.hpp:11,
                 from /home/gsaurel/fedora/robotpkg/math/pinocchio/work.fluffy64-fedora28/pinocchio-1.3.3/include/pinocchio/multibody/visitor.hpp:13,
                 from /home/gsaurel/fedora/robotpkg/math/pinocchio/work.fluffy64-fedora28/pinocchio-1.3.3/include/pinocchio/multibody/joint/joint-composite.hxx:8,
                 from /home/gsaurel/fedora/robotpkg/math/pinocchio/work.fluffy64-fedora28/pinocchio-1.3.3/include/pinocchio/multibody/joint/joint-composite.hpp:475,
                 from /home/gsaurel/fedora/robotpkg/math/pinocchio/work.fluffy64-fedora28/pinocchio-1.3.3/include/pinocchio/multibody/joint/joint-generic.hpp:10,
                 from /home/gsaurel/fedora/robotpkg/math/pinocchio/work.fluffy64-fedora28/pinocchio-1.3.3/unittest/joint-generic.cpp:5:
/usr/include/boost/fusion/container/vector/detail/deref_impl.hpp: Dans l'instanciation de « static constexpr boost::fusion::extension::deref_impl<boost::fusion::vector_iterator_tag>::apply<Iterator>::type boost::fusion::extension::deref_impl<boost::fusion::vector_iterator_tag>::apply<Iterator>::call(const Iterator&) [avec Iterator = boost::fusion::vector_iterator<boost::fusion::vector<Eigen::Matrix<double, 6, 6, 0, 6, 6>&, const bool>, 1>; boost::fusion::extension::deref_impl<boost::fusion::vector_iterator_tag>::apply<Iterator>::type = bool&] » :
/usr/include/boost/fusion/iterator/deref.hpp:63:32:   requis par « constexpr typename boost::fusion::result_of::deref<Iterator>::type boost::fusion::deref(const Iterator&) [avec Iterator = boost::fusion::vector_iterator<boost::fusion::vector<Eigen::Matrix<double, 6, 6, 0, 6, 6>&, const bool>, 1>; typename boost::fusion::result_of::deref<Iterator>::type = bool&] »
/usr/include/boost/fusion/iterator/deref.hpp:71:29:   requis par « constexpr typename boost::fusion::result_of::deref<Iterator>::type boost::fusion::operator*(const boost::fusion::iterator_base<Iterator>&) [avec Iterator = boost::fusion::vector_iterator<boost::fusion::vector<Eigen::Matrix<double, 6, 6, 0, 6, 6>&, const bool>, 1>; typename boost::fusion::result_of::deref<Iterator>::type = bool&] »
/usr/include/boost/fusion/container/vector/vector.hpp:122:41:   requis par « constexpr typename boost::fusion::vector_detail::result_of::forward_at_c<Sequence, N>::type boost::fusion::vector_detail::forward_at_c(Sequence&&) [avec int N = 1; Sequence = boost::fusion::vector<Eigen::Matrix<double, 6, 6, 0, 6, 6>&, const bool>&; typename boost::fusion::vector_detail::result_of::forward_at_c<Sequence, N>::type = bool&] »
/usr/include/boost/fusion/container/vector/vector.hpp:202:46:   requis par « constexpr boost::fusion::vector_detail::vector_data<std::integer_sequence<long unsigned int, _Idx ...>, T ...>::vector_data(boost::fusion::vector_detail::each_elem, Sequence&&) [avec Sequence = boost::fusion::vector<Eigen::Matrix<double, 6, 6, 0, 6, 6>&, const bool>&; Sequence_ = boost::fusion::vector<Eigen::Matrix<double, 6, 6, 0, 6, 6>&, const bool>; <paramètre-patron-2-3> = void; long unsigned int ...I = {0, 1}; T = {Eigen::Matrix<double, 6, 6, 0, 6, 6>&, const bool}] »
/usr/include/boost/fusion/container/vector/vector.hpp:306:75:   requis par « constexpr boost::fusion::vector<T>::vector(Sequence&&) [avec Sequence = boost::fusion::vector<Eigen::Matrix<double, 6, 6, 0, 6, 6>&, const bool>&; <paramètre-patron-2-2> = void; T = {Eigen::Matrix<double, 6, 6, 0, 6, 6>&, const bool}] »
/home/gsaurel/fedora/robotpkg/math/pinocchio/work.fluffy64-fedora28/pinocchio-1.3.3/include/pinocchio/multibody/visitor.hpp:54:83:   requis par « static ReturnType pinocchio::fusion::JointVisitorBase<JointVisitorDerived, ReturnType>::run(const pinocchio::JointModelTpl<Scalar, Options, JointCollectionTpl>&, pinocchio::JointDataTpl<Scalar, Options, JointCollectionTpl>&, ArgsTmp) [avec Scalar = double; int Options = 0; JointCollectionTpl = pinocchio::JointCollectionDefaultTpl; ArgsTmp = boost::fusion::vector<Eigen::Matrix<double, 6, 6, 0, 6, 6>&, const bool>; JointVisitorDerived = pinocchio::Join [... cut ...]
/home/gsaurel/fedora/robotpkg/math/pinocchio/work.fluffy64-fedora28/pinocchio-1.3.3/unittest/joint-generic.cpp:110:37:   requis depuis ici
/usr/include/boost/fusion/container/vector/detail/deref_impl.hpp:47:49: error: lier la référence du type « boost::fusion::extension::deref_impl<boost::fusion::vector_iterator_tag>::apply<boost::fusion::vector_iterator<boost::fusion::vector<Eigen::Matrix<double, 6, 6, 0, 6, 6>&, const bool>, 1> >::type » {aka « bool& »} à « const bool » abandonne les qualificatifs
                     return i.vec.at_impl(index());
                                                 ^
```

nim65s, 2019-01-09:
You may just be missing a few `&` here and there, if I compare #635 & #601

closed 2019-01-09 by nim65s

jcarpent, 2019-01-10:
Ok. I'm going to investigate that now.
@nim65s Thanks for this fast reporting.

jcarpent, 2019-01-10:
Fixed by #636.