stack-of-tasks/pinocchio #512: [Doc] Should we wait v2.0 to start ?

nmansard, 2018-09-10:
From @jcarpent Pinocchio 1.3.0 is already published. The next important release is 2.0.0 with codegeneration and full templatization.

So, some parts of the code can be modified, but it would be important to not work on the algo part which has a lot of modifications.

[--this post is also a test for checking how much issues can be linked with project post--]

nmansard, 2018-09-10:
Looking at https://github.com/stack-of-tasks/pinocchio/blob/topic/doc-v2/doc/outline.txt

Mostly, everything that is example-based (intro, features, tutorial) and math-based could be written. The exhaustive description of the API cannot, but that just one section, more or less. 

So I say we can write at least 50% of the doc now. Don't you think?

nmansard, 2018-09-11:
@jcarpent Can we agree on this? Would you like to spend the 10 minutes needed to label the subsections you believe cannot be documented before 2.0?

jcarpent, 2018-09-11:
What it can be done for sure without affecting the next release is:
- documenting spatial classes
- make simple examples as outer files of the documentation
- the joint base class

That must be not modified:
- algo
- model/data files
- all derived joints
- liegroups
Indeed, all of them are affected by the full templatization.

nmansard, 2018-09-11:
Let it be said.

closed 2018-09-11 by nmansard