stack-of-tasks/pinocchio #513: [Tutorials and unit tests]

gabrielebndn, 2018-09-10:
First of all: I think we all agree we want to follow the example of Eigen for the documentation. Now, in Eigen, the line between documentation and tutorial is blended. For instance, the basic functionalities of the Matrix class are exhaustively covered in a single page, which also includes some toy examples, and this is done for all other topics too. If we follow this model, then, for the proper tutorials I think we should avoid tiny examples doing the same stuff as in the doc. In my view, tutorials should be larger, more in-depth examples, such as computing the inverse kinematics and doing posture generation.

Especially if we all agree on the above plan, then I am confused about the idea of including the tutorials in the doc. Don't we already have tutorials at https://github.com/stack-of-tasks/pinocchio-tutorials ? I think we should either have all tutorials there, or all tutorials inside the doc. A compromise might be turning the tutorial repository into a subrepository of pinocchio. The point is there should not be two different sets of tutorials in two different places.

Related to the previous point: if we want to have C++ tutorials, should they be mixed to the python tutorials or be kept separate? In case, they should be at most in two different subfolders of the same folder, or two different subrepos of pinocchio. Also, what should the relationship be between the examples in the two languages? Should they mostly mirror each other by showing the same things? Should they cover different topics?

gabrielebndn, 2018-09-10:
As for the unit tests, they were in the first sketch of the documentation outline, but I am not sure about what should be said about them, nor were to put them in the structure of the documentation.
Should we make a separate part for them, like it is done for the benchmarks?

nim65s, 2018-09-10:
the content of the pinocchio-tutorials repository has been updated and moved into pinocchio's doc, right there:
https://github.com/stack-of-tasks/pinocchio/tree/master/doc/d-labs

I just archived the repository.

nim65s, 2018-09-10:
and they were more "labs" than tutorials, so we still have to write the tutorials, and we can put them there:
https://github.com/stack-of-tasks/pinocchio/tree/master/doc/c-tutorials

writing c++ and python version in separate subfolders, which could be used as tests and included in the doxygen seems to be a good idea.

jcarpent, 2018-09-10:
Just to mention that all the codes tutorials and documentation should be independent from gepetto-gui.

jcarpent, 2018-09-10:
@nim65s Why `c-tutorials` and `d-labs` naming?

gabrielebndn, 2018-09-10:
Well then, I just recommend that the old pinocchio-tutorials repository be kept visible until we get the first revision of the SII paper, as we explicitly mention it in the submitted version

gabrielebndn, 2018-09-10:
Yes, it's also not clear to me the distinction between "labs" and "tutorials"

nim65s, 2018-09-10:
I prefixed those folder with a letter, becaus doxygen includes them in an alphabetical order

The pinocchio-tutorials repo is still visible. For github, "Archived" means "read-only".

"labs" are part of a course @nmansard gives to master students. They are extensive, cover the science part, and are mainly composed of questions a studend has to work on.
" tutorials" are a more classical collection of quick examples on howto use this lib

gabrielebndn, 2018-09-10:
> The pinocchio-tutorials repo is still visible. For github, "Archived" means "read-only"

I know, I meant let's keep it this way

> "labs" are part of a course @nmansard gives to master students. They are extensive, cover the science part, and are mainly composed of questions students have to work on.

I see. So, as I said above, in my view the documentation should contain by itself some small, quick examples, the way Eigen does. My understanding was that the tutorials would be more extensive. That's why I said we should not repeat in the tutorials what has been shown in the "features" part. So, we should be careful not to solve the labs in the tutorials, otherwise it will be too easy for the students!

nmansard, 2018-09-10:
I agree with @nim65s vision: tutorials are simple piece of code with just a bit of documentation to explain it, and no math background. We may want to keep the "labs", i.e. long piece of exercices, but they should be less visible than the tutos. I would add that tutos should correspond to "killer-apps" if possible, i.e. with features that everybody wants to implement when they are using pinocchio.

About the use of examples in eigen-style documentation: I do agree that we will want examples everywhere in the doc. I like the idea that very basic examples are given as tutorials in addition to any examples in the doc. Now, if we also want to put examples in the doc outside of tutorials, should we automatically validate them? I guess yes, but we may want to keep that question for later. Let's target a first complete doc first.

jcarpent, 2018-09-10:
I guess the confusion comes from the naming.

I think the name examples or use case are the right terms, as it is already present in the examples directory of pinocchio.

And labs should be renamed practices/exercises or any equivalent term.

closed 2018-09-11 by nmansard