stack-of-tasks/pinocchio #518: [algorithm] getJacobianComFromCrba

gabrielebndn, 2018-09-14:
I took a look to getJacobianComFromCrba.
It immediately looked clear to me that this algorithm relies on the fact that the model has a floating base (exactly one floating base to be precise).
I think the same is true for getComFromCrba.
Therefore, in particular, these algorithms would fail for fixed-base manipulators. Especially the Jacobian would probably crash if the robot has less than three degrees of freedom, as data.M.topRows<3>() is called.

Is my impression correct?
Of course controlling the CoM is not as interesting for fixed-base manipulators as it is for humanoids.
But I think this restriction should be documented in the doxygen comments. Right now, I cannot find information about it anywhere.

jcarpent, 2018-09-14:
With the current CRBA version, it is true, but there is another one crbaMinimal that does not assume free floating systems.
But, getComJacobianFromCrba will be changed as soon as the official CRBA will become crbaMinimal.

gabrielebndn, 2018-09-14:
Ah, I think I see what you plan to do. Thanks!

jcarpent, 2018-09-14:
OK. I'm going to close the issue. Feel free to open it again.

closed 2018-09-14 by jcarpent