stack-of-tasks/pinocchio #523: Calling se3.aba returns "random"/unstable values

jviereck, 2018-09-20:
Hi folks,

I designed a 8 DOF quadruped URDF model and try to use pinocchio for it's simulation (the URDF model is [here](https://gist.github.com/jviereck/38aac55d69c7e14ee86cde191ceed8b0)). When I call `se3.aba(...)` after loading the URDF and creating the model/data, the returned acceloration is not stable and sometimes is `nan`:

```python
# Script test.py

import pinocchio as se3
from pinocchio.utils import zero

import numpy as np
np.set_printoptions(precision=2, linewidth=160, suppress=True)

model = se3.buildModelFromUrdf('quadroped.urdf', se3.JointModelFreeFlyer())
data = model.createData()
q, dq, tau = zero(15), zero(14), zero(8)
q[6] = 1. # Unitquarternion

print(se3.aba(model, data, q, dq, tau))
```

```bash
jviereck@xeno-ci:~/dev$ python test.py
[[      0.  ]
 [  -3529.11]
 [     -9.81]
 [ -17252.29]
 [      0.  ]
 [  -7351.28]
 [  39102.24]
 [ -22063.81]
 [ 493649.48]
 [-498318.67]
 [  39102.24]
 [ -22063.81]
 [  50600.19]
 [ -33674.3 ]]
jviereck@xeno-ci:~/dev$ python test.py
[[  0.  ]
 [ -0.09]
 [ -9.81]
 [ -0.43]
 [  0.  ]
 [ -0.18]
 [  0.98]
 [ -0.55]
 [ 12.34]
 [-12.46]
 [  0.98]
 [ -0.55]
 [  1.27]
 [ -0.84]]
```

My best guess is some data was not prepared/computed before calling `se3.aba(...)`. However, from looking at the python-bindings docstring, I don't see any hint what should be called first.

Any idea?

Best,
\jv

gabrielebndn, 2018-09-20:
I am sorry, I reopened the issue because I do not get your same behaviour on code containing a similar error.
I do not see any other problems in your code. May you try to follow my advice above and tell me what you obtain?

gabrielebndn, 2018-09-20:
Ok, I am now using the urdf you attached, and I get your same behaviour. I close the issue again. Hope my suggestion solves your problem.

closed 2018-09-20 by gabrielebndn

jcarpent, 2018-09-20:
This lines will solve the problem:
```
import pinocchio as se3 
from pinocchio.utils import zero

import numpy as np
np.set_printoptions(precision=2, linewidth=160, suppress=True)

model = se3.buildModelFromUrdf('quadroped.urdf', se3.JointModelFreeFlyer())
data = model.createData()
q, dq, tau = zero(model.nq), zero(model.nv), zero(model.nv)

# The free flyer has an idenitity placement
q[:7] = np.matrix(se3.utils.se3ToXYZQUAT(se3.SE3.Identity())).T

# or if you want to pick a random configuration
lb = model.lowerPositionLimit # lower bound from the URDF
lb[:3] = -1e3 # Translation part
lb[3:7] = -1. # Quaternion part

ub = model.upperPositionLimit # upper bound from the URDF
ub[:3] = 1e3 # Translation part
ub[3:7] = 1. # Quaternion part

q = se3.randomConfiguration(model,lb,ub) # pick between two bounds

print(se3.aba(model, data, q, dq, tau))
print data.ddq
```

jcarpent, 2018-09-20:
I also added some additional lines to allow you to exploit Pinocchio in depth (the clear documentation of the project is ongoing).

gabrielebndn, 2018-09-20:
So, @jviereck, as you can see @jcarpent  is using `model.nq` and `model.nv`. In general, this is the recommended way to use Pinocchio, rather than hard-coding 15 and 14.
Then he is showing you how to create a random configuration, but you do not need it.

Notice the line
`q[:7] = np.matrix(se3.utils.se3ToXYZQUAT(se3.SE3.Identity())).T`
In @jcarpent 's code, it is useless, since it is overwritten later. But in your code, you may want to use this line instead of `q[6] = 1.` to be more robust to errors in the index.

jviereck, 2018-09-20:
hi folks,

> I think the problem is in tau=zero(8).
> For the moment in Pinocchio all joints must be given a corresponding joint torque, INCLUDING those of the base.

Spot on - this was the issue and after changing it to `tau = zero(model.nv)`, the problem disappeared.

Thanks a lot for the quick replies & help!